cyberbotics/webots_ros2 #534: webots_ros2_driver: WEBOTS_CONTROLLER_URL seems to be ignored

ajshank, 2022-11-23:
**Bug/Issue**
When launching a driver instance of `webots_ros2_driver`, the `additional_env={'WEBOTS_CONTROLLER_URL': 'foo'}` is meant to specify the name of the robot in Webots ([ROS2 tutorial here](https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Webots.html)). However, the process works nicely even if the robot in question is named something else in the `.wbt` and `.urdf` files. Consequently, there doesn't seem to be a way to tell the driver which robot to 'connect to' when two or more robots are instantiated at launch.

**Steps to Reproduce : relevant bits**
- `robot_launch.py` file is adapted from [here](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/launch/robot_launch.py):   
   
    ```python
    turtlebot_driver = Node(
        package='webots_ros2_driver',
        executable='driver',
        output='screen',
        additional_env={'WEBOTS_CONTROLLER_URL': 'foo'},
        parameters=[
            {'robot_description': robot_description,
             'use_sim_time': use_sim_time,
             'set_robot_state_publisher': True},
            ros2_control_params
        ],
        remappings=mappings
    )
   ```

- `robot_description` comes from the default basic URDF file [here](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/resource/turtlebot_webots.urdf) (contains a different string for `name`).

- The `.wbt` file passed to Webots has this relevant section:
   ```python
   WorldInfo {
     basicTimeStep 33
   }
  TurtleBot3Burger {
    name "yoda"
    translation 0 1 0
    rotation 0 0 1 0
    controller "<extern>"
    controllerArgs [
      ""
    ]
    extensionSlot [
      RobotisLds01 {
        name "scan"
      }
    ]
  }
   ```

**Expected behavior**
The driver should fail (?), and or raise a warning. At the moment it happily connects to this robot instance.

**Affected Packages**
  - webots_ros2_driver
  - webots_ros2_turtlebot (?)

**System**
 - Webots Version: R2020a
 - ROS2 Version: Galactic/Humble 
 - Operating System: Debian/Ubuntu 20

**Additional context**
As mentioned earlier, it seems unclear how one might create multiple instances of a robot and have a driver attach to one in particular. There, we populate the `.wbt` file with two robots (different names) and create two instances of the driver (with different names passed). However, the established connections appear to be random; i.e., the controller intended for one robot sometimes connects to the other.   
Apologies if I'm missing something, or if this isn't the recommended way of doing such a thing. I'm aware of Supervisor nodes and perhaps there's a means of achieving this through that.

ad-daniel, 2022-11-24:
If the simulation only has 1 extern controller, then indeed there is no need to set the `WEBOTS_CONTROLLER_URL` variable as it will connect to the only available extern robot automatically. If there are multiple robots with extern controllers and no variable is set, then any connection would (should) be refused. So the behavior you describe is normal, but it won't be if you noticed it happening with multiple extern controllers,is that so?

ad-daniel, 2022-11-24:
So I made a few tests and it seems to behave correctly. Setting a wrong `WEBOTS_CONTROLLER_URL` is also rejected. The reason for your behavior is probably because you're using R2020a (which I assume wasn't a typo). `WEBOTS_CONTROLLER_URL` was introduced only with R2022b, before it the variable that one needs to set was called `WEBOTS_CONTROLLER_NAME`, so effectively your example is equivalent to not setting anything which indeed results in an automatic connection if only 1 robot with extern controller is present.

ajshank, 2022-11-24:
Thanks for checking, @ad-daniel.    

I assume that "multiple extern controllers" means something like so:
```python
TurtleBot3Burger {
  name "robot1"
  translation 0.16683504 -0.9041014 0
  rotation 0.0 0.0 0.999962185850965 -0.008696370976461625
  controller "<extern>"
  controllerArgs [ "" ]
  extensionSlot [ RobotisLds01 { } ]
}
TurtleBot3Burger {
  name "robot2"
  translation 0.16683504 -0.9041014 0
  rotation 0.0 0.0 0.999962185850965 -0.008696370976461625
  controller "<extern>"
  controllerArgs [ "" ]
  extensionSlot [ RobotisLds01 { } ]
}
```
with two instances of `webots_ros2_driver` (and two `diffdrive_controller` nodes). In this case, not providing any 'name' arguments to the driver will still result in _a_ driver instance connecting to _a_ robot, but not necessarily to the "correct" ones (obviously, because no names were provided). So, no connections seem to be refused in this case.

In a single robot with a single extern controller, setting the wrong name (`WEBOTS_CONTROLLER_NAME`) also has no effect: it happily connects to the only robot ("robot1") from the `.wbt` file, when this variable is set to "robot2". I can't see why it picked up the  name "robot1" to connect to, other than by _ignoring_ the provided argument and picking up the only `extern` controller found in the file.   
One additional piece of (maybe relevant) context is that the nodes for a single robot are launched under a namespace that matches the robot name. Thus, the `ros2_control_params` yaml file contains "robot1" at its root. But I don't think the driver uses/relies on that to establish a connection.

---
> If the simulation only has 1 extern controller [..] it will connect to the only available extern robot automatically.

Thanks for affirming this! I get this behaviour, although I'm not sure I've seen this explicitly in the docs. If that's correct, I'm happy to contribute.

---
PS: R2020a wasn't a typo, I was checking through an older (Foxy) documentation, mistakenly assuming that it probably applied to an older Webots version as well.

ygoumaz, 2022-12-09:
I am closing this issue for now, as this is related to an older version of Webots. Feel free to reopen if needed.

closed 2022-12-09 by ygoumaz