cyberbotics/webots_ros2 #552: webots_ros2_turtlebot slam_toolbox wrong map

zhuyeaini9, 2022-12-09:
ubuntu 22.04
ros2 humble version,after install webots-ros2 with follow:
**sudo apt-get install ros-humble-webots-ros2**
then run **ros2 launch webots_ros2_turtlebot robot_launch.py**
the webots with turtlebot run ok.
it can operate by **ros2 run teleop_twist_keyboard teleop_twist_keyboard** .
then run **ros2 launch slam_toolbox online_async_launch.py**
which install by **sudo apt install ros-humble-slam-toolbox**
run **rviz2**
after operate the turtlebot with keyboard to generate the map,
I find the map are rotated,it should can be reproduced,all packages are install with apt.
it seems have no such problem in gazebo.any advice are welcome!!!
[image]

ygoumaz, 2022-12-09:
We are currently investigating this bug with another user. There seems to be a bug in slam-toolbox with the scan matching option. Do you know if this option is activated in the config file in your case?

Moreover, we are also facing a bug on the Lidar device in Webots that may affect the correct generation of the map. I will reference this issue once the one about the Lidar is open.

closed 2023-01-05 by zhuyeaini9

toshiharutf, 2023-01-05:
Hello @zhuyeaini9,
I faced the same issue you mentioned using slam_toolbox with Webots. I observed the odometry values coming from the diff_drive node started to drift quite a lot. The documentation mentions that it requires a very precise odometry, so this could be one of the reasons. On the other hand, the documentation also mentions that the toolbox requires fine tuning of the configuration parameters, so I expect that it will not work out-of-the-box with just the default values. Did you tune those?

I got better results using cartographer: https://ros2-industrial-workshop.readthedocs.io/en/latest/_source/navigation/ROS2-Cartographer.html in Webots.

Best regards,