cyberbotics/webots_ros2 #561: URDFSpawner <device reference=”link”> does not work when the link is defined in the urdf and not in an existing PROTO

PymZoR, 2022-12-13:
**Describe the Bug**
When defining a robot using only one URDF and no existing PROTO, `reference` property of the `device` tag does not work.
In the turtlebot example, it works because the turtlebot already exists in the world file and the urdf is only used for defining webots plugins : https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/resource/turtlebot_webots.urdf#L4

**Steps to Reproduce**
Consider the following urdf : 
```xml
<?xml version="1.0"?>
<robot name="myrobot"
    xmlns:xacro="http://ros.org/wiki/xacro">

    <!-- Webots -->
    <webots>
        <device reference="laser_link" type="Lidar">
            <ros>
                <enabled>true</enabled>
                <updateRate>5</updateRate>
                <topicName>/scan</topicName>
                <alwaysOn>true</alwaysOn>
                <frameName>laser_link</frameName>
            </ros>
        </device>

        <plugin type="webots_ros2_control::Ros2Control" />
    </webots>

    <!-- BASE LINK -->
    <link name="base_link">
        <visual>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </visual>
        <collision>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </collision>
    </link>

    <joint name="laser_link_joint" type="fixed">
        <parent link="base_link" />
        <child link="laser_link" />
        <origin xyz="0 0 1" />
    </joint>

    <!-- LINK2 -->
    <link name="laser_link">
        <!-- Here the link could be empty resulting in only a Transform Node with a Lidar object, but due to issue #559
        a <collision> tag is required in order for the link to exists in webots -->
        <visual>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </visual>
        <collision>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </collision>
    </link>
</robot>
```

**NOTE** : I added `<collision>` and `<visual>` tags to the `laser_link` otherwise it is not created in webots. Probably linked to the issue : https://github.com/cyberbotics/webots_ros2/issues/559

The `laser_link` is not created, nor the Lidar object.

**Expected behavior**
A Lidar object should exist in Webots and the ROS ropic `/scan` should be advertised.

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**System**
 - Webots Version: R2023a for macOS
 - ROS Version: Humble
 - Operating System: Ubuntu jammy in docker, host macOS Monterey 12.2.1, using local_simulation_server.py

CraigMS-dev, 2023-04-01:
Was this ever solved? I have the same issue at the moment and I still haven't figured out how to get a Lidar (from any device) created from a pure urdf robot.

ygoumaz, 2023-04-04:
I am transferring this issue to Urdf2Webots repository, as it directly concerns the importer. The way to import a Lidar has been answered here: https://robotics.stackexchange.com/questions/24596/how-do-i-integrate-webots-sensor-plugins-into-a-ros2-urdf-xacro-file. The combination of gazebo and sensor tags allows to import Webots devices from URDFs and to use them with `webots_ros2` plugins.