cyberbotics/webots_ros2 #592: Can't perform the Linux installation guide for Webots ROS2 on docker.

Maxew42, 2023-01-05:
**Describe the Bug**
I need to use ROS2 and Webots in a docker container. I am starting from the Webots docker image `cyberbotics/webots` and I install ros2 on it. Both are working normally on their own side. Then I am trying to follow this guide to link webots with ROS2 : https://github.com/cyberbotics/webots_ros2/wiki/Linux-Installation-Guide
and fail at this step `ros2 launch webots_ros2_epuck robot_launch.py` with the following error `InvalidLaunchFileError: Caught exception when trying to load file of format [py]: [Errno 2] No such file or directory: 'wslpath'`

Am I missing something ? Thank you in advance.
**Steps to Reproduce**
- Build the Dockerfile presented below with docker on Windows (WSL2)
- Launch the image and access it via bash
- Follow the tutorial from https://github.com/cyberbotics/webots_ros2/wiki/Linux-Installation-Guide. Both installation  from distributed package and source are not working.

**Expected behavior**
The webots_ros2_epuck package should be running without errors.

**Affected Packages**
List of affected packages:
  - webots_ros2_epuck

**System**
 - Webots Version: cyberbotics/webots:latest
 - ROS Version: foxy
 - Operating System: Docker Ubuntu 20.04 running on Windows10
 - Graphics Card: Intel(R) UHD Graphics 620

**Aditionnal Context**
The Dockerfile used to create the image before following the ros-2 webots tutorial:
```
FROM cyberbotics/webots
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8
RUN apt update && apt install -y curl gnupg2 lsb-release git-all
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null
RUN apt update
RUN DEBIAN_FRONTEND=noninteractive apt install -y ros-foxy-desktop
WORKDIR /root/dev_ws/src
RUN git clone https://github.com/ros/ros_tutorials.git -b foxy-devel
WORKDIR /root/dev_ws
RUN apt-get install python3-rosdep -y
RUN rosdep init
RUN rosdep update
RUN rosdep install -i --from-path src --rosdistro foxy -y
RUN apt install python3-colcon-common-extensions -y
COPY ros2_entrypoint.sh /root/.
COPY scripts /home/.
ENTRYPOINT ["/root/ros2_entrypoint.sh"]
CMD ["bash"]
```

ygoumaz, 2023-01-06:
If I understand correctly, you are on a Windows machine and using a WSL environment. And what you are trying to do is install `webots_ros2` inside a Docker container started from the WSL environment? 

On Windows, the recommended approach is to directly install the package inside WSL2, without Docker (see [here](https://github.com/cyberbotics/webots_ros2/wiki/Windows-Installation-Guide)). This way, Webots is started on the host and can benefit from hardware acceleration.

Your approach may still allow hardware acceleration but it certainly is more complex, however I am not convinced that the acceleration would work using a combination of WSL and Docker on Windows. Concerning the error you are encountering, the package uses the `uname` command to determine the environment and detect if the package is started from WSL. The weird part is that you run it from inside a Linux container, which shouldn't detect the WSL environment (which seems to cause your issue). 

Could you please run the `uname` command inside your Docker container and copy the output in an answer here?

Maxew42, 2023-01-06:
Yes that’s it exactly. I would add that I am only using windows to create and test the docker image. Ultimately it will be running on Linux.

Thank you for the suggestion ! I am aware of this possibility. However, I will need to run the image in an environment where I can only run docker images and cannot install ROS2, or any other software. Therefore, installing the package in WSL is not a possibility. 

Here is the output of the uname command: `Linux`
and of the `uname -a` : `Linux d64ad31c1502 5.10.16.3-microsoft-standard-WSL2 #1 SMP Fri Apr 2 22:23:49 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

ygoumaz, 2023-01-06:
The problem is indeed here: The `uname -a` command is run inside the `webots_ros2` package and, even though we are inside the container, it returns the WSL kernel (See [this post](https://superuser.com/questions/1660970/uname-returns-host-kernel-in-docker-container) for details). This confuses the whole process of starting the Webots node.

I don't know if you are using the official `webots_ros2` released package or if you installed it from sources, but one option would be to install the package from sources and modify this line of the code: 
https://github.com/cyberbotics/webots_ros2/blob/7b6bdc0f7fb2368ebbc7fe1c31b7fc0111f5ab6a/webots_ros2_driver/webots_ros2_driver/utils.py#L88
to 
```python
return false
```
This way, the package won't use the WSL configuration and will run the same way it would behave in a Linux environment.

ygoumaz, 2023-01-06:
Glad that you were able to solve the problem!

closed 2023-01-06 by ygoumaz