cyberbotics/webots_ros2 #597: We still have this issue. I checked in multiple Ubuntu 22.04 installations for Webots 2023a. No issue when this commit is reverted.

skpawar1305, 2023-01-10:
We still have this issue. I checked in multiple Ubuntu 22.04 installations for Webots 2023a. No issue when this commit is reverted.

_Originally posted by @skpawar1305 in https://github.com/cyberbotics/webots_ros2/issues/572#issuecomment-1377990987_

ad-daniel, 2023-01-11:
Just to be sure, the referenced problem should be present ONLY when installing the webots_ros2 package from apt (and assuming the ros distro you're using was sync recently and contains the fix, you're using humble I assume?), whereas installation from source didn't pose problems. You mention reverting the commit, so can you confirm how you're installing the package? 
Also, do you get any specific error?

skpawar1305, 2023-01-11:
Broken with: recently upgraded webots_ros2 apt packages. Also with latest master branch.
Humble

ad-daniel, 2023-01-12:
Unfortunately I'm unable to reproduce the issue you're mentioning. I've tried on a clean virtual machine with R2023a + ubuntu 22 + webots_ros2 installed from sources and it works fine for me. The fact that the automatic tests work as well indicates that the issue might be on your end.

It should be noted that the last version available for humble from apt is 2023.0.0 which **does not** contain the latest fix and indeed does not work. The reason even building from source doesn't appear to work for you might be because the package from apt is installed and is being used rather than the one you've built from source. So I suggest you to try the following:

```
sudo apt remove --purge  ros-humble-webots-ros2*`
source /opt/ros/humble/setup.bash
```
then checkout the latest branch from master and 
```
colcon build && source install/setup.bash
```

skpawar1305, 2023-01-12:
launching: ros2 launch webots_ros2_tiago robot_launch.py

master branch:
[driver-6] /home/skpawar1305/Desktop/simulated_spot/install/webots_ros2_driver/lib/webots_ros2_driver/driver: error while loading shared libraries: libController.so: cannot open shared object file: No such file or directory

I checked again, no issues after reverting 
https://github.com/cyberbotics/webots_ros2/commit/241be2fb09c306350c02e5f907bf7ef9b0b8ce7e

can you upgrade that vm with sudo apt upgrade and check once? Thank you

ad-daniel, 2023-01-12:
Still working fine for me. I also asked a colleague to test on her machine and it works for her too.

https://github.com/cyberbotics/webots_ros2/commit/241be2fb09c306350c02e5f907bf7ef9b0b8ce7e is not an approach we can use because we don't have the rights to create symlinks in the build farm, that's why apt packages that use this approach don't work.

skpawar1305, 2023-01-14:
I don't know why latest apt packages and/or github master branch doesn't work for me,
anyway, I'll use 
os.environ['LD_LIBRARY_PATH'] += ':/opt/ros/humble/lib/controller' for now. Thank you.

closed 2023-01-14 by skpawar1305

ad-daniel, 2023-01-16:
Just for reference, in Humble the apt package will not working until version 2023.0.1 is released (the sync has not yet occurred). Until then, the only option is to uninstall completely the apt package and build from source.

For Rolling (for which a 2023.0.1 package is available), the apt package does work so the symlink alternative did fix the problem.

skpawar1305, 2023-02-04:
no issue again with latest apt packages, thank you