cyberbotics/webots_ros2 #603: Cannot compile `master` nor `develop` for Humble on Ubuntu 22.04

galou, 2023-01-20:
**Describe the Bug**
`colcon build --packages-up-to webots_ros2_driver` fails with the error

```
--- stderr: webots_ros2_driver
CMake Error at /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/webots_ros2_msgs__rosidl_typesupport_cExport-relwithdebinfo -dcmake_export_compile_commands=on.cmake:10 (set_target_properties):
  set_target_properties called with incorrect number of arguments.
Call Stack (most recent call first):
  /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/webots_ros2_msgs__rosidl_typesupport_cExport.cmake:68 (include)
  /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/webots_ros2_msgsConfig.cmake:41 (include)
  CMakeLists.txt:33 (find_package)

CMake Error at /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/webots_ros2_msgs__rosidl_typesupport_cppExport-relwithdebinfo -dcmake_export_compile_commands=on.cmake:10 (set_target_properties):
  set_target_properties called with incorrect number of arguments.
Call Stack (most recent call first):
  /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/webots_ros2_msgs__rosidl_typesupport_cppExport.cmake:68 (include)
  /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /home/gael/11-projects/2021-01-rur/5-work/code/robocon_ros2_monows/install/share/webots_ros2_msgs/cmake/webots_ros2_msgsConfig.cmake:41 (include)
  CMakeLists.txt:33 (find_package)

```

**Steps to Reproduce**
1. `cd $ROS_WORKSPACE`
2. `colcon build --packages-up-to webots_ros2_driver`
3. See error

I get the same error on `master` and `develop` (I ran `git submodule update --init --recursive`).

**Expected behavior**
The package should compile.

**Affected Packages**
List of affected packages:
  - webots_ros2_msgs
  - webots_ros2_driver
  - webots_ros2_control
  - webots_ros2_epuck
  - webots_ros2_tesla
  - webots_ros2_tests
  - webots_ros2_tiago
  - webots_ros2_turtlebot
  - webots_ros2_universal_robot

**System**
 - Webots Version (irrelevant, both tried):  [R2021b, R2023a]
 - ROS Version: Humble
 - Operating System: Linux Ubuntu 22.04

ad-daniel, 2023-01-20:
Odd, works for me on a VM. Did you install it following this procedure: https://github.com/cyberbotics/webots_ros2/wiki/Linux-Installation-Guide ?
In particular, did you run this part:
```
# Install dependencies
sudo apt install python3-pip python3-rosdep python3-colcon-common-extensions
sudo rosdep init && rosdep update
rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```

galou, 2023-01-20:
I just realized that the cause may have been in my call to `colcon` and maybe also combined with my default colcon configuration. I deleted my `build` directory and the problem disappeared so another explanation could be a version change in one of the tools used during the building process.
Closing the issue for loss of reproducibility.

closed 2023-01-20 by galou