cyberbotics/webots_ros2 #619: [driver-6]: process has died [pid 33255, exit code 1  when lanuch webots_ros2_turtlebot

HaoLiuHust, 2023-02-07:
**Describe the Bug**
when I lanuch the turtlebot example, there is an error:
```
[ERROR] [driver-6]: process has died [pid 33255, exit code 1, cmd '/root/ros/ros2_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_cbkel_rz --params-file /root/ros/ros2_ws/install/webots_ros2_turtlebot/share/webots_ros2_turtlebot/resource/ros2control.yml -r /diffdrive_controller/cmd_vel_unstamped:=/cmd_vel'].
```

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected Packages**
List of affected packages:
  - webots_ros2_A
  - webots_ros2_B

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Webots Version: [e.g., R2019b, R2020a revision 1]
 - ROS Version: [e.g., Dashing, Eloquent]
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

ygoumaz, 2023-02-14:
What version of the package are you using? Is it installed from sources or from the official repository? Are other examples working or crashing too?

HaoLiuHust, 2023-02-15:
I am install it from  the official repository, and tried from sources, not works...finally, I install ros2 humble, it works

closed 2023-02-15 by HaoLiuHust