cyberbotics/webots_ros2 #626: How to create a ros2 node written in c++ to co-simulate with webots？？

SenHe524, 2023-02-14:
I am trying to create a ros2 node written in c++ to co-emulate with webots. Although I have checked a lot of materials, I still have no success. I wonder if the official can write relevant tutorials?

ygoumaz, 2023-02-14:
You can create your own package for your simulations and own plugins (in C++ and Python) to extend the interface between the package and Webots. There are tutorials located in the wiki to help you:
* [Creating-a-Custom-Package](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Package) 
* [Creating-a-Custom-C++-Plugin](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Cpp-Plugin)

I am not sure to understand what your exact needs are, but in case you are looking for a more simpler node that processes data, publishes and subscribes to existing topics, you can take a look at the following tutorial on the official ROS 2 documentation: https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html.

SenHe524, 2023-02-14:
> You can create your own package for your simulations and own plugins (in C++ and Python) to extend the interface between the package and Webots. There are tutorials located in the wiki to help you: [...]

First, thank you for your reply. I have read both of the two links you provided. The first link uses python language, and the webots_ros2 function package used in the second link seems to be out of date, or I don't understand the meaning of the second function package. May I ask if there are more in-depth examples? Thanks!

ygoumaz, 2023-02-14:
In my knowledge, there is no other example based on C++. 

Concerning the tutorial on the C++ plugin, what exactly do you mean by "out of date"? Is something not working anymore? Could you give a pointer to the part that is deprecated, so I can update it?

SenHe524, 2023-02-14:
> In my knowledge, there is no other example based on C++.
> 
> Concerning the tutorial on the C++ plugin, what exactly do you mean by "out of date"? Is something not working anymore? Could you give a pointer to the part that is deprecated, so I can update it?

[image]
I tried the second linked tutorial and it reported the error shown in the image

ygoumaz, 2023-02-14:
You are right, the tutorial was broken when we switched the controller API from C++ to C in `webots_ros2_driver`.

I will fix it and let you know once it is done.

SenHe524, 2023-02-14:
> You are right, the tutorial was broken when we switched the controller API from C++ to C in `webots_ros2_driver` (#508).
> 
> I will fix it and let you know once it is done.

Ok, thank you very much. Since I am in a hurry to implement the function shown in the problem, would you mind telling me an approximate repair time, or could you give a routine under the current code framework?

ygoumaz, 2023-02-14:
This is it, the tutorial is now up-to-date and fixed. I also added additional information to make it clearer. Don't hesitate if you have any additional suggestions. Since #508, it is not possible anymore to use the C++ API from plugins. Exclusively the C API should now be used.

SenHe524, 2023-02-14:
> This is it, the tutorial is now up-to-date and fixed. I also added additional information to make it clearer. Don't hesitate if you have any additional suggestions. Since #508, it is not possible anymore to use the C++ API from plugins. Exclusively the C API should now be used.

Oh, thank you very much. I will follow the tutorial and try to write my own code. If there is any problem, I hope to wait for your help. Thank you again.

omichel, 2023-02-15:
I assume we can now close this issue. Feel free to re-open it if needed.

closed 2023-02-15 by omichel