cyberbotics/webots_ros2 #635: Drop slam-toolbox for other more robust packages

ygoumaz, 2023-02-20:
The slam-toolbox package is used in the Tiago tutorial for SLAM. There have been reports that the package can be inaccurate when used with the diff-drive controller (see https://github.com/cyberbotics/webots/issues/5540#issuecomment-1371748895). Here is the map generated with the Tiago:

[image]

As such, I suggest to replace the package with a more accurate alternative like ACLM or Cartographer in the Tiago tutorial. Cartographer is already used in the Turtlebot tutorial. @delihus has also demonstrated that ACLM gives better results, in the scope of a project using the ROSbot.

delihus, 2023-03-01:
I think you should support slam-toolbox. This is the most common and most popular mapping package in ROS2. It is also described on [navigation2_tutorials](https://github.com/ros-planning/navigation2_tutorials/).

Nevertheless I will prepare demo with the ROSbot2 and ROSbot XL with [a dockerized cartographer](https://github.com/husarion/cartographer-docker).

delihus, 2023-03-01:
There is one tricky way to get work with slam_toolbox is to disable scan_matching. But it is not satisfactory way to fix this.

I described the slam_params [here](https://github.com/husarion/rosbot-xl-mapping/blob/add-webots/config/slam_toolbox_params_webots.yaml).

There are the results:
webots world 
[image]
RViz map
[image]

ygoumaz, 2023-03-01:
Thanks, I will try this out!

lukicdarkoo, 2023-03-01:
Tiago odometry is not calibrated, it is probably the main cause of the poor mapping performance

ygoumaz, 2023-03-02:
The change to Cartographer is now merged and works fine, even without additional odometry calibration. I will try the parameters provided by @delihus with ST to see if I can get any improvements.

@lukicdarkoo, do you have any resources on how to calibrate the odometry? Should it be done using the `diff_drive_controller` package?

If I can get good results using Slam Toolbox, maybe providing the possibility to use both packages to the user would be great.

lukicdarkoo, 2023-03-03:
> @lukicdarkoo, do you have any resources on how to calibrate the odometry? Should it be done using the diff_drive_controller package?

You can do it with those two steps:
- Move the robot forward, check whether the robot overshoots or undershoots (compare the passed distance from the Odometry message and in Webots), and calibrate the diameter accordingly. Repeat this step until the Odometry and Webots distance have a big difference.
- Rotate the robot in place, check whether the robot overshoots or undershoots, and calibrate the distance between the wheels accordingly. Repeat this step until the Odometry and Webots yaw have a big difference.

The wheel distance and diameter are here:
https://github.com/cyberbotics/webots_ros2/blob/f35554ba57b02a167c72ef3240e2108ff4c0e9ad/webots_ros2_tiago/resource/ros2_control.yml#L16-L17

delihus, 2023-03-14:
Thinking back to this issue it reminded me I used [EKF](https://github.com/husarion/webots_ros2/blob/master/webots_ros2_husarion/webots_ros2_husarion/resource/ekf.yaml) `robot_localization` to create this maps. 
https://github.com/cyberbotics/webots/issues/5540#issuecomment-1348094887
It doesn't really seem like the issue is odometry.

ygoumaz, 2023-03-14:
Are you talking about the maps on this thread or in https://github.com/cyberbotics/webots/issues/5540#issuecomment-1348094887? Were you using SLAM Toolbox for the ones with the Husarion logo?

delihus, 2023-03-14:
I'm talking about this linked map https://github.com/cyberbotics/webots/issues/5540#issuecomment-1348094887 but this is the map which you prepared for ROSbot demo. This is a map of the office with the computers.
[image]

delihus, 2023-03-14:
Yes but this map is made without position correction. It is just mapping without localization.

ygoumaz, 2023-03-16:
I calibrated the TIAGo odometry and restored SLAM Toolbox in #676. It works well without scan-matching but ends up drifting after some time. Scan-matching really doesn't work, so I opened an issue on the Github repository: https://github.com/SteveMacenski/slam_toolbox/issues/586.

PR #645 (2023-03-02): Replace SLAM toolbox by Cartographer in TIAGo example
Resolves #635.

The cartographer node is responsible for the SLAM computation.
The cartographer grid node allows to display visually the map in RViz.
PR #676 (2023-03-16): Calibrate Tiago odometry and restore SLAM-Toolbox
Resolve #635.

The TIAGo odometry is calibrated to allow a better result with SLAM toolbox. 

However, the SLAM is still not 100% accurate, as the `scan_matching` option is not working correctly and is disabled in the calibration file.

The user can now choose between SLAM-toolbox and cartographer when running the TIAGo example.
closed 2023-03-16 by ygoumaz