cyberbotics/webots_ros2 #637: webots and ros2 humble connection issue

abdulkadrtr, 2023-02-21:
I am trying to establish a simple communication between Webots and ROS 2 by following the instructions in https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Webots.html, but I'm encountering issues. While Webots opens and I can see the nodes and topics with the rqt_graph tool, messages are not being transmitted. Although I can see the list of messages by using the ros2 topic list command, I am unable to listen to the messages by using the ros2 topic echo command.

**System**
 - Webots Version: R2023a
 - ROS Version: Humble 
 - Operating System: Ubuntu 22.04

ygoumaz, 2023-02-22:
Are you able to launch examples in the package? For example the one featuring universal robots?

You can launch it with:
```bash
ros2 launch webots_ros2_universal_robot multirobot_launch.py
```

Webots should be started and after a few seconds a robot should be spawned (there should be 1 orange robot and 1 one grey robot in the scene). Both should be moving.

To try out topics, you could also start the Tiago example:
```bash
ros2 launch webots_ros2_tiago robot_launch.py
```

and control the tiago with `teleop_twist_keyboard`:
```bash
ros2 run teleop_twist_keyboard teleop_twist_keyboard
```

If these are working, the issue you are encountering is probably due to something which is wrongly setup in your project.

abdulkadrtr, 2023-02-22:
Thank you for your response. multirobot_launch.py is working and the robots are moving. However, tiego robot launch is working, but teleop_twist_keyboard is not working.
[image]

ygoumaz, 2023-02-22:
If you see an output in the third terminal (the one that listens and displays the content of the /cmd_vel topic), it is likely that teleop_twist_keyboard works fine. The output that you see is simply because the key that you pressed sends the message to stop the robot. Maybe the configuration of your keyboard does not match the one given by teleop. Can you try other keys and see if some velocity values are given to the topic? You must select the terminal of teleop when pressing the keys.

abdulkadrtr, 2023-02-22:
When I ran the command `ros2 topic pub /cmd_vel geometry_msgs/Twist "{linear: {x: 0.5}}"` in the terminal, the robot started to move, so the issue was resolved. Thank you

closed 2023-02-22 by ygoumaz