cyberbotics/webots_ros2 #643: No such file or directory: 'msys64/mingw64/bin/webots.exe'

matthew-william-lock, 2023-02-24:
**Describe the Bug**
I am attempting to run `` ros2 launch webots_ros2_mavic robot_launch.py`` in order to launch the mavix example but the following error below is thrown.

```
[INFO] [launch]: All log files can be found below /home/matt/.ros/log/2023-02-24-16-52-07-526398-LAPTOP-1532NB09-17750
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [webots-1]: exception occurred while executing process:
Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/execute_local.py", line 545, in __execute_process
    transport, self._subprocess_protocol = await async_execute_process(
  File "/opt/ros/humble/lib/python3.10/site-packages/osrf_pycommon/process_utils/async_execute_process_asyncio/impl.py", line 139, in async_execute_process
    transport, protocol = await _async_execute_process_nopty(
  File "/opt/ros/humble/lib/python3.10/site-packages/osrf_pycommon/process_utils/async_execute_process_asyncio/impl.py", line 45, in _async_execute_process_nopty
    transport, protocol = await loop.subprocess_exec(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1667, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.10/asyncio/unix_events.py", line 207, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.10/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.10/asyncio/unix_events.py", line 799, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1730, in _execute_child
    self._posix_spawn(args, executable, env, restore_signals,
  File "/usr/lib/python3.10/subprocess.py", line 1675, in _posix_spawn
    self.pid = os.posix_spawn(executable, args, env, **kwargs)
FileNotFoundError: [Errno 2] No such file or directory: '/home/matt/webots/msys64/mingw64/bin/webots.exe'

[INFO] [ros2_supervisor.py-2]: process started with pid [17753]
[INFO] [driver-3]: process started with pid [17755]
[driver-3] Webots doesn't seems to be ready yet: (retry count 1)
[driver-3] Webots doesn't seems to be ready yet: (retry count 2)
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [driver-3]: process has died [pid 17755, exit code -2, cmd '/home/matt/ros2_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_kmuebe29'].
[ERROR] [ros2_supervisor.py-2]: process has died [pid 17753, exit code -2, cmd '/home/matt/ros2_ws/install/webots_ros2_driver/lib/webots_ros2_driver/ros2_supervisor.py --ros-args'].
```

**Steps to Reproduce**
1. Install webots from source
2. Clone webots_ros2 and build
3. Run `` ros2 launch webots_ros2_mavic robot_launch.py``

**System**
 - Webots Version: R2023a revision 1
 - ROS Version: Humble
 - Operating System: WSL2 Ubuntu 22.04

**Additional context**
I believe the error is occuring because I am running on WSL and have installed webots from source. To this end, I have found that https://github.com/cyberbotics/webots_ros2/blob/ca9349d9a2ed19ab6f10ba6af30e8975218e49b0/webots_ros2_driver/webots_ros2_driver/webots_launcher.py#L74

is responsible for setting the source director. I am not sure if this is the default location where webots is installed by the package manager, or if webots is attempting to launch on Windows. The latter would be helpful to improve performance of course.

matthew-william-lock, 2023-02-24:
I have tried setting WEBOTS_HOME to the installation on Windows and webots now launches, but I am still left with the following

```
[INFO] [launch]: All log files can be found below /home/matt/.ros/log/2023-02-24-18-25-51-278945-LAPTOP-1532NB09-18621
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [18623]
[INFO] [ros2_supervisor.py-2]: process started with pid [18625]
[INFO] [driver-3]: process started with pid [18627]
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 1)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 2)
[ros2_supervisor.py-2] The Webots simulation world is not ready yet, retrying for another 50 seconds...
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 3)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 4)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 5)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 6)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 7)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 8)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 9)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 10)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 11)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 12)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 13)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 14)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 15)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 16)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 17)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 18)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 19)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 20)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 21)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 22)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 23)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 24)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 25)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 26)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 27)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 28)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 29)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 30)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 31)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 32)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 33)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 34)
[driver-3] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 35)
```

ygoumaz, 2023-02-27:
When using the package on WSL, you should install Webots on Windows and set `WEBOTS_HOME` to `/mnt/c/Program\ Files/Webots` (for example). See the following instructions: https://github.com/cyberbotics/webots_ros2/wiki/Windows-Installation-Guide. You could also use your Webots installation in WSL but this is not recommanded as Webots will be very slow.

The error you are getting about `WEBOTS_SERVER` indicates that you are not using the correct version of Webots with the package. What are the versions of `webots_ros2` and Webots you are using? Also the installation of Webots in WSL may conflict in case this isn't the same version. It is recommanded to use at least Webots R2023a with the last version of the package (if you built from sources, it should be 2023.0.2).

matthew-william-lock, 2023-02-27:
Noted ! I realised it was looking for a Windows installation and have now followed your instructions. 

Please find the version information below

webots_ros2 : ca9349d9a2ed19ab6f10ba6af30e8975218e49b0
webots : Webots R2023a

ygoumaz, 2023-02-27:
This is typically a problem we encountered when users tried to use the 2023 version of the package with Webots R2022b. The version of the controller library is not the same as Webots and the controller of the Mavic is not able to connect to the instance. `WEBOTS_SERVER` is something that was removed with version R2023a.

The weird part is that the supervisor uses the correct API and is able to connect, while the Mavic one (driver - 3) was not compiled against the correct version. Here are the steps I can advise you to follow to fix the issue:

-  If, by any chance, you also have installed `webots_ros2` through apt (in addition to your sources), remove it with
    ```bash
    sudo apt remove ros-humble-webots-ros2
    ```
- Remove the version of Webots that you have installed inside WSL, because it may conflict (in particular if it is lower than R2023a) and unset `WEBOTS_HOME`.
- Inside your ROS 2 workspace, remove the three following folders: `build`, `install` and `logs`. Source ROS 2 and colcon build. This should correctly compile all executables against the correct version (including driver, which is going to start the mavic controller and connect to Webots).
    ```bash
    source /opt/ros/humble/setup.bash
    colcon build
    ```
- Source your workspace with
    ```bash
    source install/local_setup.bash
    ```

- Set `WEBOTS_HOME` to your Windows installation folder.
- Try to start the Mavic example.

In case it still doesn't work, can you please provide the full logs? Some other lines may help to understand the source of the problem.

matthew-william-lock, 2023-03-11:
Hi @ygoumaz, sorry for the delay.

I made sure to:
- remove ```webots_ros2``` through apt, which showed that nothing was installed.
- remove the version of webots inside WSL
- Remove ```build```, ```install```, ```logs``` and rebuild and source
- Set ```WEBOTS_HOME``` to my Windows installation folder.

This unfortunately did not fix the issue. What I did find is that the ```Cannot open file: /tmp/webots-1234/WEBOTS_SERVER``` was actually caused by me setting ```WEBOTS_TMPDIR``` in .bashrc as follows

```
export WEBOTS_TMPDIR=/mnt/c/Users/mtthe/AppData/Local/Temp
```

I have since removed this but still see the following error

```
$ ros2_ws$ ros2 launch webots_ros2_mavic robot_launch.py
[INFO] [launch]: All log files can be found below /home/matt/.ros/log/2023-03-11-12-36-12-620554-LAPTOP-1532NB09-3824
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [3826]
[INFO] [ros2_supervisor.py-2]: process started with pid [3828]
[INFO] [driver-3]: process started with pid [3830]
[driver-3] Webots doesn't seems to be ready yet: (retry count 1)
[driver-3] Webots doesn't seems to be ready yet: (retry count 2)
[ros2_supervisor.py-2] The Webots simulation world is not ready yet, retrying for another 50 seconds...
[driver-3] Webots doesn't seems to be ready yet: (retry count 3)
[driver-3] Webots doesn't seems to be ready yet: (retry count 4)
[driver-3] Webots doesn't seems to be ready yet: (retry count 5)
[driver-3] Webots doesn't seems to be ready yet: (retry count 6)
[driver-3] Webots doesn't seems to be ready yet: (retry count 7)
[driver-3] Webots doesn't seems to be ready yet: (retry count 8)
[ros2_supervisor.py-2] The Webots simulation world is not ready yet, retrying for another 45 seconds...
[driver-3] Webots doesn't seems to be ready yet: (retry count 9)
[driver-3] Webots doesn't seems to be ready yet: (retry count 10)
[driver-3] Webots doesn't seems to be ready yet: (retry count 11)
[driver-3] Webots doesn't seems to be ready yet: (retry count 12)
```

By full logs, do you mean the content of ```launch.log```?

matthew-william-lock, 2023-03-11:
I have just done a clean install on WSL and it seems to be working...

omichel, 2023-03-13:
I am closing this issue as it seems to be resolved. Feel free to re-open if needed.

closed 2023-03-13 by omichel