cyberbotics/webots_ros2 #647: ROS2_WEBOTS_HOME error

abdulkadrtr, 2023-02-28:
I installed webots_ros2 and when I run the command `ros2 launch webots_ros2_mavic robot_launch.py` directly in the terminal, everything works correctly. Then, I created a ros2_ws folder and copied the "webots_ros2_mavic" folder from the webots_ros2 repository to the "ros2_ws/src" directory. After that, the `colcon build` command runs correctly. However, when I source the 
`source install/setup.bash` file inside the "ros2_ws" folder and run the `ros2 launch webots_ros2_mavic robot_launch.py` command, it does not work and gives the following error. What I am trying to do is to run the "mavic" package in the "ros2_ws" folder.

[INFO] [launch]: All log files can be found below /home/abd/.ros/log/2023-02-28-22-59-20-513563-abd-22365
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /home/abd/.ros/webotsR2023a/webots.
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: Action.__init__() got an unexpected keyword argument 'ros2_supervisor'

Also, running the command echo `export ROS2_WEBOTS_HOME=/opt/ros/humble/share/webots_ros2/">> ~/.bashrc` doesn't seem to work.

System

Webots Version: R2023a
ROS Version: Humble
Operating System: Ubuntu 22.04

ygoumaz, 2023-03-01:
There is one warning and one error here.

- `Warning`: The warning tells you that you haven't set either `ROS2_WEBOTS_HOME` or `WEBOTS_HOME`. In this case, the package will try to start Webots from the default installation directory (`usr/local/webots`). You can set, for example, `WEBOTS_HOME` to a custom installation folder.
- `Error`: The way you are trying to start the Mavic from sources by simply copying the folder won't work. This sub-package is dependent to other sub-packages of `webots_ros2`, like `webots_ros2_driver` for example. Please follow the instructions on the following page (the ones about sources installation) for a complete and clean workspace: https://github.com/cyberbotics/webots_ros2/wiki/Linux-Installation-Guide#install-webots_ros2-from-sources. I would also recommend to remove the package you installed using apt, because it could conflict with the one in your `ros2_ws`.

abdulkadrtr, 2023-03-01:
Yes, it works now in the ros2_ws workspace. Thanks.

closed 2023-03-01 by ygoumaz