cyberbotics/webots_ros2 #654: Mavic Example not Running with Source Build on Foxy

frapit, 2023-03-02:
**Describe the Bug**
I cant get the mavic example run with ubuntu20.04 and ros foxy. The node crashes with 
```
[ERROR] [driver-3]: process has died [pid 632170, exit code 1, cmd '/home/user/ws/ros2/dev_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_tmzp5qug'].

```

**Steps to Reproduce**
- Install webots
- Clone webots_ros2 repo
- run `ros2 launch webots_ros2_mavic robot_launch.py`

**Expected behavior**
The drone to fly.

**System**
 - Webots Version: R2023a
 - ROS Version: FOXY
 - Operating System: ubuntu20
 - Graphics Card: Nvidia T-2000

I thought the git repo tag 2023.0.2 would be compatible with the webots version. The default demo (gripper) does work. The binary, version does also work, but I couldn't get any source build to work. When checking out 2022.x tags, I get the following building issues:

```
Starting >>> webots_ros2_driver
--- stderr: webots_ros2_driver                         
CMake Error at /opt/ros/foxy/share/ament_cmake_python/cmake/ament_python_install_package.cmake:46 (message):
  ament_python_install_package() the Python package folder
  '/home/user/ws/ros2/dev_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/python38'
  doesn't contain an '__init__.py' file
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_python/cmake/ament_python_install_package.cmake:28 (_ament_cmake_python_install_package)
  CMakeLists.txt:108 (ament_python_install_package)
```

ad-daniel, 2023-03-02:
Did you install it following these [instructions](https://github.com/cyberbotics/webots_ros2/wiki/Complete-Installation-Guide#installation-instructions)?

ygoumaz, 2023-03-02:
> ```
> Starting >>> webots_ros2_driver
> --- stderr: webots_ros2_driver
> CMake Error at /opt/ros/foxy/share/ament_cmake_python/cmake/ament_python_install_package.cmake:46 [...]

From the error you have given, I can see that the package you are trying to build is not up-to-date. The following line is from an old version of the package.
```
src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/python38
```
Currently in the sources, it should be
```
src/webots_ros2/webots_ros2_driver/webots/lib/controller/python
```

frapit, 2023-03-02:
Thanks for the quick help. That is what I wrote, when I check out the up-to-date repo, the driver crashes at runtime. I then checked out earlier versions (to reproduce the behaviour of ros-foxy-webots-ros2) when those build errors you cited came up.

ygoumaz, 2023-03-02:
Can you provide the complete logs when the driver crashes with the 2023.0.2 version? It could be a problem of controller library version.

frapit, 2023-03-02:
Sure, here are the logs from the 2023.0.2 version but I couldn't find any stack trace or debug mode to know why the driver crashed:
```
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-03-02-15-49-32-885248-nuxor-663903
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /usr/local/webots.
[INFO] [webots-1]: process started with pid [663905]
[INFO] [ros2_supervisor.py-2]: process started with pid [663907]
[INFO] [driver-3]: process started with pid [663909]
[driver-3] Cannot connect to Webots instance, retrying in 0 second...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 50 seconds...
[driver-3] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Mavic_2_PRO/extern", retrying in 1 second...
[driver-3] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Mavic_2_PRO/extern", retrying in 2 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 45 seconds...
[driver-3] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Mavic_2_PRO/extern", retrying in 3 seconds...
[ERROR] [driver-3]: process has died [pid 663909, exit code 1, cmd '/home/user/oe124/workspaces/ros2/dev_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_erm4132o'].

```
 whereas in Webots I get the following logs:

```
INFO: 'Mavic_2_PRO' extern controller: waiting for connection on ipc://1234/Mavic_2_PRO or on tcp://<ip_address>:1234/Mavic_2_PRO
INFO: 'Ros2Supervisor' extern controller: waiting for connection on ipc://1234/Ros2Supervisor or on tcp://<ip_address>:1234/Ros2Supervisor
INFO: 'Ros2Supervisor' extern controller: connected.
INFO: 'Mavic_2_PRO' extern controller: connected.
INFO: 'Mavic_2_PRO' extern controller: refusing connection attempt from another extern controller.
INFO: 'Mavic_2_PRO' extern controller: disconnected, waiting for new connection.
```

ygoumaz, 2023-03-02:
The driver logs (the countdown especially) is from a previous version of the controller library. The compilation may have gone wrong.

To be sure everything works fine, remove the apt package of `webots_ros2`, because it could interfer with the sources package. Then remove the three directories in your workspace: `build`, `install` and `logs`. Then, try to rebuild the package with `colcon build`.

When running the example the driver logs should be like this:
```
[driver-6] Cannot connect to Webots instance, retrying for another 50 seconds...
[driver-6] Cannot connect to Webots instance, retrying for another 45 seconds...
```

frapit, 2023-03-02:
Thank you, I thought I already removed the binary package but apparently I removed only the meta package `apt-get remove ros-foxy-webots-ros2` and not the controller. After rebuilding and removing the distro version, it works perferctly. Thanks for the quick support!

closed 2023-03-02 by frapit