cyberbotics/webots_ros2 #659: Missing force torque sensor implementation in webots_ros2_control

borongyuan, 2023-03-06:
Hi,
I found that there are configurations about force torque sensor in the urdf file of universal robot. But it seems that there is no corresponding implementation in webots_ros2_control. Do you have any plans to implement this part for use with [force_torque_sensor_broadcaster](https://github.com/ros-controls/ros2_controllers/tree/master/force_torque_sensor_broadcaster)?
Thanks
https://github.com/cyberbotics/webots_ros2/blob/f35554ba57b02a167c72ef3240e2108ff4c0e9ad/webots_ros2_universal_robot/resource/ur5e_with_gripper.urdf#L78-L85

ad-daniel, 2023-03-08:
Hi, thank you for the report. We don't have such plans in the short term, but feel free to open a pull request and propose its implementation, we'll be happy to review it

borongyuan, 2023-03-10:
This is more complicated than I expected. I took a quick look through the code and found that this part is missing in webots. Among the [Sensors](https://cyberbotics.com/doc/guide/sensors) of Webots, only [TouchSensor](https://cyberbotics.com/doc/reference/touchsensor) can provide 3D force, but it still lacks 3D torque. I will study this part later when I have time, if the issue still exists at that time.

omichel, 2023-03-10:
Yes, you should probably create a new type of touch sensor that includes the torque feedback in addition to the force feedback...

ygoumaz, 2023-04-12:
I am closing this, as we won't implement it.

closed 2023-04-12 by ygoumaz