cyberbotics/webots_ros2 #660: Problems encountered during simulation

SenHe524, 2023-03-06:
When I tried to use the c++ function package written by myself to conduct co-simulation with webots in ros2, my terminal reported an error as shown in the following figure:
[image]
But my simulation can still run normally, may I ask how this is going on, and how should I modify, so as not to report errors

ad-daniel, 2023-03-08:
You mention the simulation still runs normally, so I assume eventually it stops printing this message, right? In that case the message is to be expected. Essentially the controller (ROS2Supervisor) attempts to connect to the webots instance, but this is not possible until the webots world is fully loaded. If the world is complex, this might take some time. 
If you don't need ROS2supervisor, you can also disable it in the latest version by setting `ros2_supervisor=False` in the WebotsLauncher (more details here https://github.com/cyberbotics/webots_ros2/pull/616)

SenHe524, 2023-03-08:
The error is always present and does not disappear over time, and during this time, the simulation runs normally. My simulation world is really complicated, I don't know if that's the reason.

ad-daniel, 2023-03-08:
What I meant is, when the simulation begins to run, the error should disappear. It should be there only during the loading process.
It doesn't seem like you need it though, so perhaps the best option is just to disable it

SenHe524, 2023-03-08:
When I shut down ROS2Supervisor, the error is no longer reported, but will shutting down ROS2Supervisor have any effect on the simulation? I'm new to webots, so please forgive me for asking silly questions!

SenHe524, 2023-03-08:
Ok, after I closed the ROS2Supervisor node according to your suggestion, the terminal no longer reported cyclic errors, but there was still a connection failure at the beginning. The error picture is as follows:
[image]
My launch file is as follows:
[image]

ygoumaz, 2023-03-13:
Let me provide you with a detailed explanation of the error you have been experiencing.

Firstly, I noticed that in the launcher file you provided, the `WebotsLauncher` (named `webots`) is created without the `ros_supervisor` parameter. As mentioned by @ad-daniel , this parameter is False by default, which means that no Supervisor node is added to the world file, and `Ros2SupervisorLauncher` is not automatically called by the Webots-ROS2 interface, resulting in the absence of the corresponding ROS 2 node. This is not an issue in your case as you don't require the Supervisor node's ability to spawn URDF or Webots node on the fly.

Secondly, the initial error you encountered regarding the `Ros2Supervisor` attempting to connect despite the simulation running correctly, was due to you manually creating the `Ros2SupervisorLauncher` (named `ros2_supervisor`) object inside your launch file. However, this has been prohibited since the package's version 2023.0.2. Instead, it is recommended that you always use the `ros_supervisor` parameter of the `WebotsLauncher` by setting it to true, allowing the interface to create all the necessary ROS2 and Webots nodes automatically. Consequently, you can remove the commented lines from your launch file.

Lastly, the last INFO message you received from `driver-2` is normal and indicates that your plugin is attempting to connect to the Robot of the world while Webots is loading. These messages are expected and will continue to display until the simulation is ready.

SenHe524, 2023-03-13:
I see. Thank you very much for your reply.

> Let me provide you with a detailed explanation of the error you have been experiencing.
>
> Firstly, I noticed that in the launcher file you provided, the `WebotsLauncher` (named `webots`) is created [...]