cyberbotics/webots_ros2 #668: the data type of the topic “point cloud”

warmsh, 2023-03-10:
**Documentation page(s) concerned**
[Link to the concerned page.](https://www.cyberbotics.com/doc/reference/lidar?tab-language=ros#wb_lidar_enable_point_cloud)

**What is the problem with the current documentation?**
According to the official manual, the data type of the  topic “point cloud” published by lidar in webots is sensor_msgs::PointCloud, but how should I modify the data type of sensor_msgs::PointCloud2?

[image]

ad-daniel, 2023-03-10:
The documentation page your reference is about ROS, but this is the ROS2 repo. Which of the two are you using?
Concerning ROS2, the point cloud publisher uses [PointCloud2](https://github.com/cyberbotics/webots_ros2/blob/ba1ad64436328037c0de3d5dbb2834ece2ec6a4b/webots_ros2_driver/src/plugins/static/Ros2Lidar.cpp#L48)

ad-daniel, 2023-03-13:
I will close this for now, feel free to re-open it if needed

closed 2023-03-13 by ad-daniel