cyberbotics/webots_ros2 #669: Launching of complex world fails. Webots closed while loading

yarens1991, 2023-03-11:
I have been trying to modify the TIAGo slam tutorial to adapt it to more complex scenarios. The first modification I tried involved >650 MB .obj files and it was successful. However, when I make the scenario more complex (>2.5 GB .obj files) Webots is closed during initialization. I am able to load this scenario directly from Webots, the problem appears just when is launched from webots_ros2. Due to the complexity of the scenario it takes long time to load, so I suspect there is a timeout preventing Webots to complete the initialization. I would like to know if there is a way to pass a parameter to make the launching process waiting longer for Webots? I have already tried modifying the controller_manager_timeout parameter in robot_launch.py but it doesn't have any effect.

**Steps to Reproduce**
1. Modify webots_ros2_tiago/worlds/default.wbt (insert >2.5 GB .obj files)
2. Launch the example

**Expected behavior**
Simulation launched successfully 

**Terminal output**
[INFO] [launch]: All log files can be found below /home/xymbot/.ros/log/2023-03-11-01-14-13-003342-xymbot-OMEN-by-HP-Laptop-16-b1xxx-25553
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /home/xymbot/.ros/webotsR2023a/webots.
[INFO] [webots-1]: process started with pid [25554]
[INFO] [ros2_supervisor.py-2]: process started with pid [25556]
[INFO] [spawner-3]: process started with pid [25560]
[INFO] [spawner-4]: process started with pid [25562]
[INFO] [robot_state_publisher-5]: process started with pid [25564]
[INFO] [driver-6]: process started with pid [25567]
[INFO] [static_transform_publisher-7]: process started with pid [25569]
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 50 seconds...
[static_transform_publisher-7] [WARN] [1678493653.185923781] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-7] [INFO] [1678493653.193588383] [static_transform_publisher_8TJEACL8Sm965J5M]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'base_link' to 'base_footprint'
[robot_state_publisher-5] [INFO] [1678493653.198092560] [robot_state_publisher]: got segment 
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 50 seconds...
[spawner-3] [INFO] [1678493655.388423488] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493655.388613562] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493657.395214819] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493657.395276260] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 45 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 45 seconds...
[spawner-3] [INFO] [1678493659.401995355] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493659.401994201] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493661.408633094] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493661.408635697] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 40 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 40 seconds...
[spawner-3] [INFO] [1678493663.415620619] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493663.415622576] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493665.422464129] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493665.422464151] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493667.429328729] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493667.429329043] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 35 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 35 seconds...
[spawner-3] [INFO] [1678493669.436008101] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493669.436006531] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493671.442100794] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493671.442280182] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 30 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 30 seconds...
[spawner-3] [INFO] [1678493673.448817635] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493673.448817816] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493675.455622804] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493675.455622087] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493677.462108026] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493677.462162265] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 25 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 25 seconds...
[spawner-4] [INFO] [1678493679.468500803] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493679.468499195] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493681.474852975] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493681.474853283] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 20 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 20 seconds...
[spawner-4] [INFO] [1678493683.481270134] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493683.481331715] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[... 353 lines cut ...]
[spawner-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 182, in main
[spawner-3]     if not wait_for_controller_manager(node, controller_manager_name,
[spawner-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 100, in wait_for_controller_manager
[spawner-3]     node_and_namespace = wait_for_value_or(
[... cut ...]
[ERROR] [spawner-4]: process has died [pid 25562, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner diffdrive_controller --controller-manager-timeout 500 --ros-args'].
[ERROR] [spawner-3]: process has died [pid 25560, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager-timeout 500 --ros-args'].

**System**
 - Webots Version: [R2023a]
 - ROS Version: [Humble]
 - Operating System: [Linux Ubuntu 22.04]
 - Graphics Card: [NVIDIA GeForce RTX 3070 6 GB]

ygoumaz, 2023-03-13:
Based on the logs, it seems that the controllers are crashing due to a timeout during the loading of the Webots simulation.

Webots controllers have a hardcoded timeout of 50 seconds for connecting to simulations. If a simulation takes longer than this time, the controllers will give up and crash, as you have experienced. Therefore, I recommend reducing the size of your .obj file to make it more manageable for Webots. This should improve the loading time and prevent the controllers from crashing due to the timeout. It is really not recommended to use very large assets files in Webots.

Another (hacky and not recommended) way to fix the issue is to modify the `reset_handler` in the launch file of the TIAGo example. The handler restarts all nodes when the supervisor exits. This explains the logs you have provided but is not the desired behavior as it duplicates a lot of nodes once the controllers crash. To prevent this, you could modify the `reset_handler` to only restart the controller nodes (driver and supervisor) and not all nodes, such as the navigation and `diff_drive_controller` nodes. This way, the controllers will simply restart and continue trying to connect to the simulation once they timeout. However this will break the initial purpose of the handler.

yarens1991, 2023-03-14:
Thank you for the quick reply @ygoumaz.

The controllers seems to be crashing after 50 seconds due to the timeout although I modified this value to 500 (see this value in the log: [ERROR] [spawner-3]: process has died [pid 25560, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager-timeout 500 --ros-args'].)

However, the controllers are automatically respawned twice before Webots crashes. In fact, when Webots crashes the third instance of the controllers is still running and it remains 25 seconds to give up connecting to the simulation. The crash always occurs around 125 seconds after launching. Apparently Webots process is being killed after this time and it generates an error.

[driver-16] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Tiago_Base/extern", retrying for another 25 seconds...
[spawner-9] [INFO] [1678493780.984805395] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1678493780.984924483] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 25 seconds...
[spawner-3] [INFO] [1678493781.787356905] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493781.800276049] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-13] [INFO] [1678493782.211549099] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-14] [INFO] [1678493782.211738892] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1678493783.015266700] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1678493783.015385120] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1678493783.819553237] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1678493783.819563037] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-13] [INFO] [1678493784.299749616] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-14] [INFO] [1678493784.299754435] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[webots-1] /home/xymbot/.ros/webotsR2023a/webots/webots: línea 102: 25566 Terminado (killed) "$webots_home/bin/webots-bin" "$@"
[ERROR] [webots-1]: process has died [pid 25554, exit code 137, cmd '/home/xymbot/.ros/webotsR2023a/webots/webots /tmp/tmp7rn4ziqw_world_with_URDF_robot.wbt --batch --mode=realtime'].

Since I can open the world directly from Webots, I'm assuming there are not timeouts in Webots and the operative system is not the responsible for killing the process. It only happens when launched from webots_ros2. I have checked the WebotsLauncher definition, but I couldn't find any parameters for controlling the timeout during Webots initialization.