cyberbotics/webots_ros2 #675: ros2 and webots did not connect with each in tutotial "Setting up a robot simulation (Webots)"

zilong-ding, 2023-03-15:
[image]
I follow the tutotial step by step,and launch the ros2 launch file,and then pub the topic as below 
[image]
but the webots has no response,I donot know why?
because the examples on webots_ros2 run correctly

ygoumaz, 2023-03-15:
On the screen you sent, the supervisor doesn't seem to be connected, preventing the simulation from running. Are you talking about the tutorial in the official documentation of ROS2? In the tutorial, we removed the Ros2Supervisor, so it shouldn't be present in your Webots world.

zilong-ding, 2023-03-16:
yes,it is the tutorial in the ros2-foxy official documentation,and i do follow the tutotial without any changes.
I open the my_world.wbt by webots open world,and the Ros2Supervisor did not appear,then I relaunch the launch file,and find the webots opens a world file in the dir tmp,and in the end of this file,I found Ros2Supervisor,but my_world.wbt does not have  Ros2Supervisor,it seems that when webotlaunch loads my_world.wbt,it automaticly adds the ROS2Supervisor

zilong-ding, 2023-03-16:
I read the code of webots turtlebot,and I found it's launch file has ros2supervisor,and I change my my_package launch file likes below
[image]
and finally,it moves!

ygoumaz, 2023-03-16:
By adding the `ROS2SupervisorLauncher()`, you indeed spawn the node that connects to the supervisor in the world and it allows to simulation to run properly. However this is not in the tutorial, because we changed the way the supervisor is spawned. This means that you may not have the latest version of the package. Be sure to install `webots_ros2` **2023.0.2** to benefit from all latest features.

closed 2023-03-16 by ygoumaz