cyberbotics/webots_ros2 #679: Accelerometer x-axes is always 0.049

zilong-ding, 2023-03-21:
I run turtlebots in webots model for slam, and when I echo the imu msg ,I found x accelerometer is always 0.049,even when I launch the project, it's x value is 0.049, is it normal or not?

ygoumaz, 2023-03-21:
I tested on my side and observed that the Turtlebot's front ball joint (sphere) collision with the floor relies on the world's basic time step. This problem causes the Turtlebot to sink slightly into the ground in the front, resulting in a slight tilt and a non-zero x value in the accelerometer due to gravity.

To resolve this issue, one solution is to move the robot around and take a series of measurements to compute the mean value. This mean value can then be subtracted from all future accelerometer measurements to eliminate the effect of the ball joint and ensure more accurate readings.

omichel, 2023-03-21:
Another solution would be to slightly change the rotation of the accelerometer of the Turtlebot so that it is perfectly vertical and hence returns only some positive value on the Z axis.

zilong-ding, 2023-03-22:
I tried the two method,but it don't work very well.anyway thanks for your opinions

closed 2023-03-22 by ygoumaz