cyberbotics/webots_ros2 #684: TurtleBot3Burger' extern controller: refusing connection attempt from another extern controller

zilong-ding, 2023-03-24:
ros2 version is foxy
webots is r2022b and r2023a

I do not know why it happens,it run normal in past few days.
[image]
I run the command of 'ros2 launch webots_ros2_turtlebot robot_launch.py '

and the output is :

[INFO] [launch]: All log files can be found below /home/xzb/.ros/log/2023-03-24-14-42-34-105438-xzb-ThinkBook-56558
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [56562]
[INFO] [ros2_supervisor.py-2]: process started with pid [56564]
[INFO] [spawner.py-3]: process started with pid [56568]
[INFO] [spawner.py-4]: process started with pid [56570]
[INFO] [robot_state_publisher-5]: process started with pid [56572]
[INFO] [driver-6]: process started with pid [56574]
[INFO] [static_transform_publisher-7]: process started with pid [56577]
[driver-6] Cannot connect to Webots instance, retrying in 0 second...
[robot_state_publisher-5] Parsing robot urdf xml string.
[robot_state_publisher-5] [INFO] [1679640154.180160885] [robot_state_publisher]: got segment 
[static_transform_publisher-7] [INFO] [1679640154.186256413] [static_transform_publisher_jmbyi35X4dqklOJj]: Spinning until killed publishing transform from 'base_link' to 'base_footprint'
[spawner.py-3] [INFO] [1679640154.298463307] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-4] [INFO] [1679640154.313857788] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[ros2_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/Ros2Supervisor/extern", retrying for another 50 seconds...
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/TurtleBot3Burger/extern", retrying in 1 second...
[spawner.py-3] [INFO] [1679640156.305413860] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-4] [INFO] [1679640156.320405710] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[driver-6] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/TurtleBot3Burger/extern", retrying in 2 seconds...
[spawner.py-3] [INFO] [1679640158.319136436] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-4] [INFO] [1679640158.332091899] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[robot_state_publisher-5] Parsing robot urdf xml string.
[robot_state_publisher-5] Link LDS-01 had 6 children
[robot_state_publisher-5] Link lower had 0 children
[robot_state_publisher-5] Link screw0 had 0 children
[robot_state_publisher-5] Link screw1 had 0 children
[robot_state_publisher-5] Link screw2 had 0 children
[robot_state_publisher-5] Link screw3 had 0 children
[robot_state_publisher-5] Link solid had 0 children
[robot_state_publisher-5] Link accelerometer had 0 children
[robot_state_publisher-5] Link compass had 0 children
[robot_state_publisher-5] Link gps had 0 children
[robot_state_publisher-5] Link gyro had 0 children
[robot_state_publisher-5] Link imu_link had 0 children
[robot_state_publisher-5] Link inertial_unit had 0 children
[robot_state_publisher-5] Link left wheel had 0 children
[robot_state_publisher-5] Link right wheel had 0 children
[robot_state_publisher-5] [INFO] [1679640160.272247721] [robot_state_publisher]: got segment LDS-01
[robot_state_publisher-5] [INFO] [1679640160.272287784] [robot_state_publisher]: got segment accelerometer
[robot_state_publisher-5] [INFO] [1679640160.272298393] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1679640160.272306343] [robot_state_publisher]: got segment compass
[robot_state_publisher-5] [INFO] [1679640160.272313800] [robot_state_publisher]: got segment gps
[robot_state_publisher-5] [INFO] [1679640160.272320978] [robot_state_publisher]: got segment gyro
[robot_state_publisher-5] [INFO] [1679640160.272327855] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-5] [INFO] [1679640160.272334649] [robot_state_publisher]: got segment inertial_unit
[robot_state_publisher-5] [INFO] [1679640160.272341577] [robot_state_publisher]: got segment left wheel
[robot_state_publisher-5] [INFO] [1679640160.272348522] [robot_state_publisher]: got segment lower
[robot_state_publisher-5] [INFO] [1679640160.272355133] [robot_state_publisher]: got segment right wheel
[robot_state_publisher-5] [INFO] [1679640160.272361933] [robot_state_publisher]: got segment screw0
[robot_state_publisher-5] [INFO] [1679640160.272368724] [robot_state_publisher]: got segment screw1
[robot_state_publisher-5] [INFO] [1679640160.272375476] [robot_state_publisher]: got segment screw2
[robot_state_publisher-5] [INFO] [1679640160.272382257] [robot_state_publisher]: got segment screw3
[robot_state_publisher-5] [INFO] [1679640160.272389156] [robot_state_publisher]: got segment solid
[ERROR] [driver-6]: process has died [pid 56574, exit code 1, cmd '/home/xzb/ros2-project/webots-ros2/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_8e8cfe67 --params-file /home/xzb/ros2-project/webots-ros2/install/webots_ros2_turtlebot/share/webots_ros2_turtlebot/resource/ros2control.yml -r /diffdrive_controller/cmd_vel_unstamped:=/cmd_vel'].

ygoumaz, 2023-03-24:
You mention 2 versions of Webots (R2022b and R2023a). When using the package from sources, it is possible that multiple Webots versions interfere with the package. The current version of the package is not compatible with R2022b. I would suggest to remove this installation and remove the `build`, `install` and `log` directories in /home/xzb/ros2-project/webots-ros2. You can then freshly build the package with `colcon build`.

ygoumaz, 2023-03-24:
What version of the package are you using? Are you on the master branch? If so, which commit are you trying to build?

zilong-ding, 2023-03-24:
Unpacking ros-foxy-webots-ros2-turtlebot (2023.0.2-1focal.20230317.032511)
2023.0.2 foxy version

ygoumaz, 2023-03-24:
From your logs the package seems to be installed from sources, but your last message references a binary package installation. Try to open a new terminal and **don't source** `/home/xzb/ros2-project/webots-ros2/install/local_setup.bash`.

This should use the binary installation by default.

ygoumaz, 2023-03-24:
Try a complete clean installation. Remove all Webots installations, binary and sources versions of `webots_ros2` and perform a fresh installation of `webots_ros2` and Webots R2023a. The error you encounter is typically due to conflicts between multiple installations of both frameworks.

zilong-ding, 2023-03-24:
actually,i tried that removing apt installed package,and just running the source code built.and already tried that removing source code,reinstall apt package,and run it. both worked not well.because in yesterday i only have apt installed package in my computer,and worked well,today it got error

ygoumaz, 2023-03-24:
`ros-foxy-webots-ros2` with Webots R2023a works well on my side on Ubuntu 20.04. The issue is certainly coming from your configuration.

zilong-ding, 2023-03-25:
i reinstall ros-foxy, and sudo apt install ros-foxy-webots-ros2.

when running "ros2 launch webots_ros2_universal_robot multirobot_launch.py "
[image]
works well
so is epuck,tiago

when running "ros2 launch webots_ros2_turtlebot robot_launch.py"
[image]
works error
so is mavic,tesla,

so coufused!

zilong-ding, 2023-03-25:
i give up trying. and use humble docker and same ros domain id to use webots, it works

bresilla, 2023-04-01:
I am having the same exact issue.  When I create a custom driver in `urdf` file: `<plugin type="webots_ros2_pioneer3at.plugin_example.PluginExample" \>` then i get the same error. 

Investigating in the code i saw that both "mavic" and "tesla" have custom drivers. 

The same goes for "turtlebot". In fact turtlebot has `<plugin type="webots_ros2_control::Ros2Control" />` as main controller, but in addition it has the example plugin too: `<plugin type="webots_ros2_turtlebot.plugin_example.PluginExample" \>`.

The same issue i think is reported #619 too

zilong-ding, 2023-04-03:
> I am having the same exact issue. When I create a custom driver in `urdf` file: `<plugin type="webots_ros2_pioneer3at.plugin_example.PluginExample" />` then i get the same error.
> 
> Investigating in the code i saw that both "mavic" and "tesla" have custom drivers.
> 
> The same goes for "turtlebot". In fact turtlebot has `<plugin type="webots_ros2_control::Ros2Control" />` as main controller, but in addition it has the example plugin too: `<plugin type="webots_ros2_turtlebot.plugin_example.PluginExample" />`. Once I comment it out the "custom driver" then it launches successfully.
> 
> The same issue i think is reported in #619 too.

yes,it worked,thanks.it has confused me some days

closed 2023-07-26 by ygoumaz