cyberbotics/webots_ros2 #704: [driver-2] Warning:"xxx" device not found.

cdlycode, 2023-04-10:
hello, I want to build my robot and run it, but I got an error after configuring according to the official example!

Webots: R2022a
ROS2: rolling
Ubuntu: 20.04

[driver-2] Warning:"xxx" device not found.
[driver-2] Warning:"xxx" device not found.
[ERROR] [driver-2] : process has died [pid xx, exit code -11, cmd '/opt/ros/rolling/lib/webots_ros2_driver/driiver --ros-args --params-file xx --params-file xxx.yaml'].

launch, yaml, urdf... these files are the same as the official examples, I don't know what the problem is?

cdlycode, 2023-04-11:
My webots version is R2022a, I can you start simple examples. I try to up-to-date version of Webots, this error can be avoided, but it seems controller_manager service cannot start, causing my trajectory controller to also fail to start.
    Then I return the version to Webots R2022a, I can avoid this error(device not found) by adding environment variables('WEBOTS_ROBOT_NAME') to the 'webots_ros2_driver' node of the launch.py file, but I get another problem...
    My trajectory controller will continue to block during the configuration process without any exception. May I ask what is the reason why the spawner script configure controller keeps getting stuck.

ygoumaz, 2023-04-11:
Could you please provide your launch file so that I can take a look? What version of the `webots_ros2` package are you using? You can check the version with:
```bash
ros2 pkg xml rviz2 | grep version
```

cdlycode, 2023-04-13:
I'm sorry I can't provide my launch file, the version of the `webots_ros2` package is 1.2.2. 

By analyzing the source code, we have identified that the problem lies in `switch_controller` in `spawner.py` script. After the client carries the request to the call server, the process is stuck at `rclpy.spin_until_future_complete`, I don't know why this is.

Notably, I have another similar robot that works very well! These two robots have the same configuration except for different `.wbt` model files, so I think there may be some issues with my `wbt` file, I would like to know if there are any factors in the `wbt` file that affect controller startup. Thank you very much for your answer!

ygoumaz, 2023-04-13:
The way controllers connect to Webots from ROS 2 (and in general) has evolved a lot in the past year. Version 1.2.2 of the package is old too, but should be compatible with Webots R2022a. There is a lot of incompatiblity factors that can come into play because of version mismatch. For these versions, you should theoretically define WEBOTS_CONTROLLER_NAME environment variable in your launch file to define which robot to connect to.

The best advice I can give to you is to update Webots to the latest stable release (R2023a) and update the `webots_ros2` package to 2023.0.2 (or higher). This way you will have the up-to-date interface to get the controller working with Webots. All tutorials and resources are now given for these versions. You can check the tutorial for basics here: https://docs.ros.org/en/rolling/Tutorials/Advanced/Simulators/Webots.html.

Additionally, `.wbt` files represent Webots worlds and not robots. Robots files are usually defined as PROTOs (`.proto`). The robot controller should be set to `<extern>`. If the robot is correctly shown in the world, it should not be a problem with the .proto file.

ygoumaz, 2023-04-17:
What is not working in our second model? Does it appear in the Webots simulation? If yes, is it interfacing with ROS 2? If not, could you please share the complete logs that you get in the Webots console and the ones from ROS 2?

Without the PROTO models, it is difficult for me to know where the problem is coming from.

closed 2023-07-26 by ygoumaz