cyberbotics/webots_ros2 #708: AsyncParametersClient gets stuck in get parameter

fafux, 2023-04-14:
**Describe the Bug**
Hi all,
I launch a Webots simulation with the joint state broadcaster and everything works.
Once I try to load my own controller, this gets stuck in the function "on_configure()" in which it tries to get a parameter. If I try to get the same parameter from the command line it works. Below, you can find the part of the code in which the controller gets stuck.
```
std::shared_ptr<rclcpp::AsyncParametersClient> parameters_client = std::make_shared<rclcpp::AsyncParametersClient>(get_node(), "/robot_state_publisher");
parameters_client->wait_for_service();
auto param = parameters_client->get_parameters({"robot_description"}).get();
```
It gets stuck in ".get()". The same code works without Webots.

**Affected Packages**
List of affected packages:
  - webots_ros2

**System**
 - Webots Version: R2023a
 - ROS Version: Foxy
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce GTX 1650

ygoumaz, 2023-04-17:
Getting a parameter value should work even if Webots is running. I don't know what your get_node() function returns but the problem may come from there. It probably gets stuck because of the get() function which is waiting for a value that is never coming.

An alternative code that should work is the following:

```cpp
  std::string robot_description;
  if (node->get_parameter("robot_description", robot_description)) {
    RCLCPP_INFO(node->get_logger(), "Robot description: %s",
                robot_description.c_str());
  } else {
    RCLCPP_ERROR(node->get_logger(), "Failed to get robot_model parameter");
  }
```

fafux, 2023-04-17:
In this way it works but it's possible to do in this way only if the parameter is read from the node that created the parameter. If I want to get a parameter of another node I should declare and use a parameter client. With a parameter client, the execution gets stuck. If I try to do the same from command line I get the parameter. So the parameter exists and is set.

ygoumaz, 2023-04-17:
You are right. The following code works on my side and is non-blocking for Webots. I checked and it is indeed the `robot_description` of the `robot_state_publisher` that is returned.

```cpp
  std::shared_ptr<rclcpp::AsyncParametersClient> parameters_client = std::make_shared<rclcpp::AsyncParametersClient>(node, "robot_state_publisher");

  auto parameter_future = parameters_client->get_parameters({"robot_description"});
  rclcpp::spin_until_future_complete(node->get_node_base_interface(), parameter_future);

  auto param = parameter_future.get().at(0);
  RCLCPP_INFO(node->get_logger(), "Got global param: %s", param.as_string().c_str());
```

fafux, 2023-04-17:
The code does the same that I provided before. The code launching webots gets stuck, without works.

ygoumaz, 2023-04-17:
There is probably something else wrong with your launcher or plugin code. Could you please share both files, so that I can further investigate the problem?

fafux, 2023-04-18:
Adding `rclcpp::spin_until_future_complete(node->get_node_base_interface(), parameter_future);` line to my code works. The problem is that in a ros2 controller you cannot use spin_until_future_complete because you get the runtime error `what():  Node has already been added to an executor.`. If you try to use [wait()](https://en.cppreference.com/w/cpp/thread/future/wait) method of std::future type it waits forever. If I try to get the parameter from the command line, while the future is waiting, I can get the parameter. The strange thing is that, with the same code, without Webots, this does not happen. I think that the key difference is on the controller manager.

ygoumaz, 2023-04-18:
From your description, I have the feeling that the restriction you encounter are coming from ros2_control packages and are not related to Webots. On my side, I am able to run Webots and get the parameter without any problem or error message. You may open an issue on https://github.com/ros-controls/ros2_control if the problem is located in the [controller manager](https://control.ros.org/foxy/doc/ros2_control/controller_manager/doc/userdoc.html).

fafux, 2023-04-18:
I do not think that is a ros2 control problem since without Webots works.

ygoumaz, 2023-04-18:
Using `rclcpp::spin_until_future_complete(node->get_node_base_interface(), parameter_future);` in a custom plugin doesn't result in the following error `what():  Node has already been added to an executor.` when I try to use it. Also, I can get the parameter correctly and the Webots simulation runs smoothly. By the way, I am on Humble, it may be a problem in Foxy.

closed 2023-05-03 by ygoumaz