cyberbotics/webots_ros2 #730: The GPU doesn't seem to be in use

944284742, 2023-05-05:
1.First I use the mirror image:  docker pull nvidia/cuda:11.1.1-cudnn8-devel-ubuntu20.04,Executing the nvidia-smi instruction is valid。So I should have no problem with the graphics driver

2.Then I set up the ROS2 and webots-ROS2 environment, and run the command: ros2 launch webots_ros2_epuck robot_launch.py。The operation was successful, but the arm was very slow and I suspect it was not using the GPU properly

3.My GPU model is Tesla P40, I wonder if this can be supported，In addition to perform glxinfo | grep OpenGL instructions to print is as follows:

root@726637f37136:/my_work/webots_ros2_ws# glxinfo | grep OpenGL
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 12.0.0, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 21.2.6
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

Summary: May I ask whether there is a problem with my environment or webot-ROS2 does not support the deployment of video card container

omichel, 2023-05-05:
From the `glxinfo` output, it seems you are not using the NVIDIA accelerated OpenGL, but the Mesa one.

944284742, 2023-05-05:
> From the `glxinfo` output, it seems you are not using the NVIDIA accelerated OpenGL, but the Mesa one.
My GPU is NVIDIA，Why is OpenGL not used? Is there something wrong with my OpenGL environment?

omichel, 2023-05-05:
To use NVIDIA hardware acceleration in a Docker environment, you should follow a special procedure documented by NVIDIA. See documentation [here](https://www.cyberbotics.com/doc/guide/installation-procedure?version=develop#run-webots-in-docker-with-gui).

closed 2023-05-16 by ygoumaz