cyberbotics/webots_ros2 #734: Slam Point Cloud Discontinuity and V-Shaped Gap in Scan

animesh-singhal, 2023-05-08:
**Bug Description**
The slam point cloud exhibits an unexpected behavior when the robot passes through the center of the room. Despite the absence of any obstacles, a V-shaped gap appears in the scan, disrupting the continuity of the point cloud. This issue is also reflected in the slam implementation, as the map fails to develop accurately within the V-shaped void.

**Steps to Reproduce**

1) Visit the following page for reference: [Example-TIAGo](https://github.com/cyberbotics/webots_ros2/wiki/Example-TIAGo)
2) Set up the webots_ros2 package. If you choose to install via binaries, please note that a few dependencies will not be automatically installed, which can result in errors during the colcon build process.
3) Install the ros-humble-slam-toolbox package by running the following command: sudo apt install ros-humble-slam-toolbox.
4) Launch the scene using the following command: ros2 launch webots_ros2_tiago robot_launch.py rviz:=true slam_toolbox:=true.

**Affected Packages**
List of affected packages:
  - webots_ros2_tiago

**Screenshots**
JIF and screenshot attached
[image]
[image]
[image]
[image]

**System**
 - Webots Version: [latest]
 - ROS Version: [Humble]
 - Operating System: [Ubuntu Jammy]

ygoumaz, 2023-05-09:
The V-shape that you see is completely normal. The light blue semicircle that you can see is the maximal range of the Lidar. If there is no object between the robot and this maximal range, then no point is detected.

animesh-singhal, 2023-05-09:
Thanks a lot for clarifying this!

animesh-singhal, 2023-05-11:
The presence of a V-shaped gap in the point cloud can significantly impact the results of the SLAM algorithm. In particular, this gap can result in unexplored areas being displayed in Rviz, which may not accurately represent the true environment. While these gaps may eventually be filled in with white space as the robot moves forward, this is not necessarily the desired behavior. It is not desirable to have a V-shaped gap in the point cloud, as seen in the Webots simulator.

[image]
[image]

ygoumaz, 2023-05-12:
As mentioned previously, the observed behavior is normal. In the SLAM Toolbox, the costmap is updated exclusively upon detecting obstacles. By increasing the range, the occurrence of V-shaped gaps can be significantly reduced.

In Webots, we display the range without any ray visualization when no obstacles are detected. While I'm not very familiar with the inner workings of Gazebo, it is likely that the ray visualization is purely for visual purposes, while the data transmitted through ROS 2 topics follows a similar approach to Webots.

To gain further insight into the behavior within Gazebo, you can check the following part of a video tutorial on interfacing Gazebo and SLAM-Toolbox (https://youtu.be/JXnXnAXrYj8?t=811). The same V-shaped areas can be observed.

It's possible that there might be a SLAM Toolbox parameter related to the cost map generation that allows it to be generated even when no points are given and that I am not aware of. I would rather ask directly on their repository if this is the case.

closed 2023-05-16 by ygoumaz