cyberbotics/webots_ros2 #741: Issues with TurtleBot Example: URDF Export, Joint Velocity Limit, and Rviz Failure

animesh-singhal, 2023-05-10:
**Bug Description**
When executing the turtlebot example from [this](https://github.com/cyberbotics/webots_ros2/wiki/SLAM-with-TurtleBot3) link, I encountered several issues after running the command:: 
`ros2 launch webots_ros2_turtlebot robot_launch.py mapper:=true rviz:=true`

Issue#1: Upon spawning, the webots console displays the following warnings:

    WARNING: Cannot initialize the sound engine: Cannot initialize OpenAL default device 'OpenAL Soft'.
    WARNING: TurtleBot3Burger (PROTO) > DEF CASTER_JOINT BallJoint: Exporting 'BallJoint' nodes to URDF is currently not supported.

Issue#2: While attempting to move the robot using telelop testing, the webots console shows the following warnings:

    WARNING: TurtleBot3Burger (PROTO) > DEF RIGHT_JOINT HingeJoint > RotationalMotor: The requested velocity 15.1515 exceeds 'maxVelocity' = 6.67.
    WARNING: TurtleBot3Burger (PROTO) > DEF LEFT_JOINT HingeJoint > RotationalMotor: The requested velocity 15.1515 exceeds 'maxVelocity' = 6.67.

Issue#3: Rviz fails to launch, preventing SLAM functionality. Additionally, enabling Optional Rendering for point clouds in webots does not display the point clouds.

**Steps to Reproduce**
Install the webots_ros2 package via binaries. 

**Affected Packages**
List of affected packages:
    webots_ros2_turtlebot

**System**
    Webots Version: [latest]
    ROS Version: [Humble]
    Operating System: [Ubuntu Jammy]

ygoumaz, 2023-05-12:
Thank you for reporting this. We are currently working on updating those tutorial pages and this one was still referring to an old version of the package. I updated the launch commands. You should now be able to use the Turtlebot example correctly with SLAM and Navigation tools.

closed 2023-05-12 by ygoumaz

animesh-singhal, 2023-05-15:
Sounds good. Thanks!