cyberbotics/webots_ros2 #747: Issue with colcon build on webots_ros2_driver

jainishmehta, 2023-05-15:
**Describe the Bug**
A clear and concise description of what the bug is.
I get this when I run colcon build. I am using ubuntu 20 focal.

--- stderr: webots_ros2_driver                                
/usr/bin/ld: warning: libconsole_bridge.so.1.0, needed by /home/jainish/ros2_foxy/ros2-linux/lib/libclass_loader.so, may conflict with libconsole_bridge.so.0.4
/usr/bin/ld: CMakeFiles/driver.dir/src/WebotsNode.cpp.o: in function `boost::filesystem::operator/(boost::filesystem::path const&, boost::filesystem::path const&)':
WebotsNode.cpp:(.text._ZN5boost10filesystemdvERKNS0_4pathES3_[_ZN5boost10filesystemdvERKNS0_4pathES3_]+0x4a): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/driver.dir/src/WebotsNode.cpp.o: in function `boost::filesystem::exists(boost::filesystem::path const&)':
WebotsNode.cpp:(.text._ZN5boost10filesystem6existsERKNS0_4pathE[_ZN5boost10filesystem6existsERKNS0_4pathE]+0x33): undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/usr/bin/ld: CMakeFiles/driver.dir/src/WebotsNode.cpp.o: in function `pluginlib::ClassLoader<webots_ros2_driver::PluginInterface>::ClassLoader(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)':
WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x10b): undefined reference to `ros::console::g_initialized'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x11b): undefined reference to `ros::console::initialize()'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x1a6): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x1fd): undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x209): undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x27a): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x298): undefined reference to `ros::package::getPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x3e7): undefined reference to `ros::console::g_initialized'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x3f7): undefined reference to `ros::console::initialize()'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x482): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x4d9): undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x4e5): undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEEC5ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE]+0x55a): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)'
/usr/bin/ld: CMakeFiles/driver.dir/src/WebotsNode.cpp.o: in function `pluginlib::ClassLoader<webots_ros2_driver::PluginInterface>::~ClassLoader()':
WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEED2Ev[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEED5Ev]+0x39): undefined reference to `ros::console::g_initialized'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEED2Ev[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEED5Ev]+0x49): undefined reference to `ros::console::initialize()'
/usr/bin/ld: WebotsNode.cpp:(.text._ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEED2Ev[_ZN9pluginlib11ClassLoaderIN18webots_ros2_driver15PluginInterfaceEED5Ev]+0xce): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocat [... cut ...]
A clear and concise description of what you expected to happen.

**Affected Packages**
List of affected packages:
  - webots_ros2_A
  - webots_ros2_B

**Screenshots**
I run colcon build to get this:
[image]
I changed up some lines of code in Ros2Camera.cpp  

[image]

**System**
 - Webots Version: [e.g., R2019b, R2020a revision 1]
 - ROS Version: Noetic (i think)
 - Operating System: [ ubuntu 20.04
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

jainishmehta, 2023-05-15:
I just want it working, so even sudo apt-get install ros-$ROS_DISTRO-webots-ros2 does not work on my system as it says cannot find the module:
[image]

ygoumaz, 2023-05-16:
First of all, Noetic is a ROS 1 distro, while `webots_ros2` is for ROS 2. This may be the first big issue.

The stderr you get while building the package seems to be related to a multiple distribution conflict between your ROS installations. Do you have multiple ROS distributions installed? You can try to install a ROS 2 distribution and set your `ROS_VERSION` and `ROS_DISTRO` variables correctly.

Note: you specify that you are using Webots R2020a. The current ROS 2 package won't work with this version. You should use Webots R2023a instead.

jainishmehta, 2023-05-16:
sudo apt install ros-${ROS_DISTRO}-ros2-control
This command does not work.
[image]

jainishmehta, 2023-05-17:
[image]
I redid it and it seems to be installed correctly, but same error for ros2_control not being found. I am not sure if others can reproduce the same bug in their system.

destogl, 2023-05-18:
You have to use foxy branch for Foxy. Please check the readme file for ditto compatibility.

ygoumaz, 2023-05-22:
I am closing this one for now, feel free to reopen if needed.

closed 2023-05-22 by ygoumaz