cyberbotics/webots_ros2 #748: libhardware_interface.so Issue when colcon build for webots_ros2_control

jainishmehta, 2023-05-16:
**Describe the Bug**
[image]
I am trying to do colcon build for webots_ros2_control.  I did git clone from ros2_control to create a package on ros2 for hardware interface and I get this wierd error similar to https://github.com/ros/class_loader/issues/188 . 

**Steps to Reproduce**
1. Clone from https://github.com/ros-controls/ros2_control for webots_ros2_control. 
2. Do colcon build to get the packages for it.

**Expected behavior**
Should work fine and build properly.

**Screenshots**
[image]

**System**
 - Webots Version: R2023
 - ROS Version: 2, FOXY (Not sure abut foxy though)
 - Operating System: Ubuntu 20.04 focal

jainishmehta, 2023-05-16:
HI, This is the issue from webots_ros2_control.
[image]

ygoumaz, 2023-05-16:
You definitely have a conflict between multiple ROS installations. The issue here is not related to `webots_ros2` but to the `ros2_control` package. I suggest to first clarify what your ROS distro is and install basic packages like `ros2_control` using apt.
```
sudo apt install ros-${ROS_DISTRO}-ros2-control
```

Please do not open multiple issues, you can directly post on #747.

closed 2023-05-16 by ygoumaz

jainishmehta, 2023-05-16:
Hi, 

[image]

The command does not work

jainishmehta, 2023-05-17:
[image]
Also, sorry I redid it and it is already installed and still the same error when I do colcon build. I agree the error is in ros2_control, however I can't find a workaround. NOt sure if people can reproduce this bug.