cyberbotics/webots_ros2 #749: Ros2Camera publishes non-canonical topics and camera info

justinberi, 2023-05-17:
**Describe the Bug**
The `mCameraInfoMessage.header.stamp` member is not updated when publishing successive camera_info messages in the `Ros2Camera` static plugin. This means the `message_filters` package cannot synchronise image and camera_info topics. According to the [message api](https://docs.ros2.org/latest/api/sensor_msgs/msg/CameraInfo.html) the stamp should be 

```
#std_msgs/Header header # Header timestamp should be acquisition time of image
```

I suggest adding to `void Ros2Camera::publishImage()`
```
mCameraInfoMessage.header.stamp = mImageMessage.header.stamp;
```
Further, the image topic is named `mTopicName` which resolves to `cameraname_color` when the simulation is run. _In my case this is `/UR5e/D405_color` (for the camera info topic it is `/UR5e/D405_color/camera_info`) when created through a [Kinect Rangefinder Sensor](https://cyberbotics.com/doc/guide/range-finder-sensors)._ According to the [message api](https://docs.ros2.org/latest/api/sensor_msgs/msg/CameraInfo.html) the topic should be one of 

```
# image_raw - raw data from the camera driver, possibly Bayer encoded

# image - monochrome, distorted

# image_color - color, distorted

# image_rect - monochrome, rectified

# image_rect_color - color, rectified
```

I suggest the following change to the plugin to indicate a "color, distorted" image.

```
// Image publisher
mImagePublisher = mNode->create_publisher<sensor_msgs::msg::Image>(mTopicName + "/image_color", rclcpp::SensorDataQoS().reliable());
```

I am happy to implement both these changes if someone could confirm they are correct. (I'm not a perception person).

**Steps to Reproduce**
Add a camera to a sim and observe ros2 topics.

(Is there simple example?)

**Expected behavior**
* camera_info topic provides the header stamp as the image acquisition time.
* image topic is named `image_color` relative to it's namespace.

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**Screenshots**
[image]

**System**
 - Webots Version: R2020a
 - ROS Version: humble
 - Operating System: Ubuntu 22.04

ygoumaz, 2023-05-17:
Thanks for reporting these issues. You can go ahead with the implementation and we will be happy to review the PR.

I suggest to target the develop branch for the fix, as we already updated some plugins (including the camera). Concerning the camera image topic name, we should modify the default value of `mImageSuffix`: 

https://github.com/cyberbotics/webots_ros2/blob/2bd2ac1bda3e5839ac55c8984eaa2f9e9f1bbd43/webots_ros2_driver/src/plugins/static/Ros2Camera.cpp#L27

justinberi, 2023-06-01:
> Thanks for reporting these issues. You can go ahead with the implementation and we will be happy to review the PR.
>
> I suggest to target the develop branch for the fix, as we already updated some [...]

Unfortunately the deveop branch is not working for me on humble.
```
justin@anvil:~/Documents/forge$ ros2 launch webots_ros2_universal_robot multirobot_launch.py
[INFO] [launch]: All log files can be found below /home/justin/.ros/log/2023-06-01-21-52-35-042174-anvil-23506
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /usr/local/webots.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [webots-1]: process started with pid [23507]
[INFO] [ros2_supervisor.py-2]: process started with pid [23509]
[INFO] [ros2-3]: process started with pid [23513]
[INFO] [webots_controller_abbirb4600-4]: process started with pid [23515]
[webots_controller_abbirb4600-4] Cannot connect to Webots instance, retrying for another 50 seconds...
[ros2_supervisor.py-2] Cannot find any instance of Webots, retrying for another 50 seconds...
[webots_controller_abbirb4600-4] Cannot connect to Webots instance, retrying for another 45 seconds...
[ros2_supervisor.py-2] Cannot find any instance of Webots, retrying for another 45 seconds...
[webots_controller_abbirb4600-4] Cannot connect to Webots instance, retrying for another 40 seconds...
[ros2_supervisor.py-2] Cannot find any instance of Webots, retrying for another 40 seconds...
[webots_controller_abbirb4600-4] Cannot connect to Webots instance, retrying for another 35 seconds...
[ros2_supervisor.py-2] Cannot find any instance of Webots, retrying for another 35 seconds...

```

ygoumaz, 2023-06-01:
The develop branch must be used with the develop version of Webots, as new changes in the controller library are not compatible with the last release of Webots.

You can install the last nightly build from this link: https://github.com/cyberbotics/webots/releases/tag/nightly_31_5_2023. Choose the R2023b version for your platform.

PR #761 (2023-06-02): 749 ros2camera publishes non canonical topics and camera info
closes #749
closed 2023-06-02 by justinberi