cyberbotics/webots_ros2 #759: No external controller connection(ROS2 on WLS to Webots on Windows)

Dan010123, 2023-05-25:
**Describe the Bug**
launched ROS2 Webot demos open Webots, show the world but stuck on waiting for extern controller connection on IPC or TCP which newer happens.

**Steps to Reproduce**
1. Installed ROS2  + ros-humble-webots-ros2 on WSL Ubuntu 22.04
2. Installed Webots on Windows10
3. Configured workspace, sourced ROS, exported WEBOTS_HOME=/mnt/d/Programs/Webots
4. Started  (for example) ros2 launch webots_ros2_turtlebot robot_launch.py

**Expected behavior**
Simulation running

**Screenshots**
[image]
[image]

**System**
 - WIndows 10 + Ubuntu 22.04 on WSL

**Additional context**
I guess I should somehow get IPC or TCP configured, but can't find info on how to manage this....
Thanks!

ygoumaz, 2023-05-25:
I will take a look at your issue. It is possible that the package is somehow broken on Windows.

Could you just give me a few more details:

- Have you built the ROS 2 package (webots_ros2) from sources or installed from apt binaries?
- What version of the package are you using? You can execute the following command to get it: `ros2 pkg xml webots_ros2 | grep version`.

suresh-venkate, 2023-05-25:
> I will take a look at your issue. It is possible that the package is somehow broken on Windows.
>
> Could you just give me a few more details:
>
> * Have you built the ROS 2 package (webots_ros2) from [...]

In my case, I have installed from apt binaries using this command: 'sudo apt-get install ros-$ROS_DISTRO-webots-ros2'. (ROS_DISTRO is humble). 

The package version that I see is ' <version>2023.0.3</version>'

Thanks

ygoumaz, 2023-05-25:
I am not able to reproduce the issue. I tried the package from sources and apt binaries and both work in version 2023.0.3 in WSL (Humble). Could you please try to install it from sources by following these instructions: https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Webots/Installation-Ubuntu.html. Remove the apt package to avoid conflicts. The current version of sources is 2023.0.4, which is released and should be synchronized with Humble and Iron very soon.

The weird part in the logs provided by @Dan010123, is that the driver and supervisor nodes are started but they don't produce any logs and don't crash. They should write messages like the following one:
```
[driver-4] The Webots simulation world is not ready yet, retrying for another 50 seconds...
```

Are there any other logs related to these nodes?

ygoumaz, 2023-05-25:
Your controllers are started correctly, but there seems to be a problem preventing the extern controllers from connecting to the simulation.

When starting a controller node from the launch file, the `WEBOTS_CONTROLLER_URL` environment variable must be set to the target the correct instance of Webots and the correct robot.

Now that you have the package installed from the sources, could you please open the launch file of any project (for example the epuck: `ros2_ws/src/webots_ros2/webots_ros2_epuck/launch/robot_launch.py`) and print the value of `controller_url_prefix()`? You can add the following line:
```python
print(controller_url_prefix())
```

Could you also try to start the mavic example? It is even simpler than the epuck and turtlebot. This would allow to discard potential issues coming from other packages like ros_control.

suresh-venkate, 2023-05-25:
I tried the mavic example. I am again unable to communicate with the webots window using teleop_twist_keyboard. 

I see this now in the webots console.

[image]

Regarding the print statement, can you tell me where exactly in the robot_launch.py file, I need to add this command? Should I add it below the import statements at the top or inside one of the functions defined in the launch.py file?

ygoumaz, 2023-05-26:
You can add the following line anywhere in the `get_ros2_nodes(*args)` function.
```python
print('prefix: ' + controller_url_prefix())
```

When running the launch file, you should get a line in the console with the prefix.

suresh-venkate, 2023-05-26:
This is the output on the console with the print statement included.

[image]

ygoumaz, 2023-05-26:
In all the logs you've provided, the counter doesn't go below 50 or 45 seconds. Can you confirm that it is still looking for the Webots instance?

According to the output of the prefix, the packet correctly detects the WSL environment and tries to connect via TCP connection. There may be a problem with the firewall somehow preventing the connection. You can check if the Webots instance is detected by WSL by following these steps:

- Start Webots on the host.
- Intall nmap in WSL:
```
sudo apt install nmap
```

- In WSL, scan for open ports:
```
nmap 172.18.112.1 -Pn
```

Is the 1234 port detected?

In addition, you can also try the following command to test the TCP connection (with Webots still open):
```
telnet 172.18.112.1 1234
```

Dan010123, 2023-05-26:
For me the prefix was tcp://172.18.224.1:1234/
Independent whether Webots is running or not, the port 1234 isn't opened with all firewalls being off
With nmap 172.18.224.1 -Pn  I have only
PORT     STATE SERVICE
1042/tcp open  afrog
1043/tcp open  boinc
Maybe :1234 should be somehow allowed in WSL/Ubuntu? And how do I know that Webots are listening as 172.18.224.1:1234 on Windows?

Dan010123, 2023-05-28:
Seems I've found the reason.
1. in Windows Defender Firewall I allowed all inbound connections on UDP/TCP to _D:\programs\webots\msys64\mingw64\bin\webots-bin.exe_ (source folder for Webots is D:/Programs/ in my case)
2. in cmd(or Pwrshell) with adm privileges 
**netstat -ab** 
    showed that port 1234 was used  even when Webots was not running (in my case - by iphlpsvc)
**netsh interface portproxy show all** 
    showed that 1234 was used namely by netsh
**netsh interface portproxy delete v4tov4 listenport=1234 listenaddress=0.0.0.0**
 removed netsh from 1234

3. now  **netstat -ab**  shows the port 1234 is not used
4.  start the Webot, **netstat -ab** to make sure that 1234 is used only by webots-bin.exe
5. with Webots running, from terminal in WSL Ubuntu  **telnet 172.18.224.1 1234** (172.18... - the ip of Ethernet adapter vEthernet (WSL), the one that was returned by "prefix" from Launch.py) - it should connect
6. if not - the Firewall settings should be checked, for example, 

[image]

may be vEthernet(WSL) should be checked-off (in my case it isn't and it doesn't affect)

michaelperino, 2023-05-31:
I fought this issue for a while and it was windows firewall on my end. When you have the firewall popup and click "allow for private networks", it creates a matching rule which blocks for public networks. Disabling the matching block for public networks and extending the original inbound rule to work for both private and public networks was the solution for me.

ygoumaz, 2023-05-31:
It indeed seems to be a firewall issue. @Dan010123, were you able to launch your `webots_ros2` examples after your fix? If so, I think it would be great to open a new post on https://robotics.stackexchange.com/ with the `webots` and `ros2` tags. I think there is a way to post a Q&A style question where you can provide the question and the answer at the same time. This could then be used as a reference for future users who encounter this problem.

ygoumaz, 2023-07-26:
@Dan010123, @suresh-venkate, any updates on this issue? Are you able to run demos now? If not feel free to reopen the issue.

closed 2023-07-26 by ygoumaz

nanjo712, 2023-11-29:
hey, I meet the same problem in webots R2023b with Ubuntu 22.04 on WSL 2.0.11.
It seems that the latest WSL has the same network enviroment as the Host. But my ROS humble still cannot connect to Webots instance.

nanjo712, 2023-11-29:
[image]
if I use NAT mode in WSL ,it will start to listen on port 1234. But ROS still cannot connect to webots.
If I use mirrored mode in WSL, it will tell me that it cannot open the port 1234 , using 1235 instead.

maze-debug, 2024-01-05:
@nanjo712  hi i have the same issue has you and can't find any solution did you succeed to fix the problem ?

cjmess01, 2024-08-29:
I was experiencing the same problem as above. I was using WSL 2.2.0. **I uninstalled wsl and reinstalled version 2.1.5.0, and that solved the issue for me.**

mstuettgen, 2024-10-21:
I was using WSL version: 2.3.24.0 and ran into the same issues. Downgrading to WSL 2.1.5.0 solved the issue for me as well.

myudak, 2025-04-24:
is there any work without downgrading, im still having the same issues