cyberbotics/webots_ros2 #765: Ros2SensorPlugin enable and disable convention.

justinberi, 2023-06-07:
**Is your feature request related to a problem? Please describe.**
There does not seem to be a way to manually enable and disable many of the currently implemented `Ros2SensorPlugin`s. 

For example the `Ros2Camera` device is currently enabled if there is a subscription to specific topics or if `alwaysOn` was set in the urdf.
https://github.com/cyberbotics/webots_ros2/blob/b8fe29e3afd77ae425e11962a98b11d052768861/webots_ros2_driver/src/plugins/static/Ros2Camera.cpp#L92
There are instances where we would like to toggle these devices on/off whilst still maintaining a subscription to them. 

Conversely, `Ros2LED` is enabled/disabled via an Int32 message on a topic defined therein.

It would be good to have consistent and documented behaviour in these plugins to avoid confusion.

**Describe the solution you'd like**
Add a subscription to a desired "enabled" state topic. 

* (Optional for clarity) Deprecate `alwaysOn` parameter in urdf.
* (Optional for clarity) Add `enabled` parameter in urdf. If `true` it starts as enabled but may be disabled. If `false` it starts disabled but may be enabled. Default is `true`.
* Add `/TOPIC_NAME/enabled` subscription to `Ros2SensorPlugin` and create abstract method `virtual void onEnabledRecieved(...) = 0` to enforce each sensor to implement.

**Describe alternatives you've considered**
To maintain backwards compatibility, `alwaysOn` could be retained and topic subscription and logic could be implemented specific to each child class of `Ros2SensorPlugin`.

**Additional context**

ygoumaz, 2023-06-26:
I agree that the wiki page of devices https://github.com/cyberbotics/webots_ros2/wiki/References-Devices should be enhanced to list all available parameters for each of them. 

However, I am not sure about deprecating the `alwaysOn` parameter. As mentioned by @lukicdarkoo [here](https://github.com/cyberbotics/webots_ros2/pull/677#discussion_r1144428628), enable and disable topics/services are never used in real hardware. Also Gazebo uses the same tag to specify if a device should be enabled or not. For consistency, I would let it the way it currently is.

justinberi, 2023-07-03:
Closing as it has been considered and the outcome is to leave as is.

closed 2023-07-03 by justinberi