cyberbotics/webots_ros2 #785: Problem using rclpy.spin_until_future_complete inside Python plugin

angel-ayala, 2023-06-23:
I'm using the Emitter send_string service call inside the `step()` python method, and when using `rclpy.spin_until_future_complete()` occurs a deadlock since the future never gets done. Apparently is because the `step()` never ends and therefore, the driver step never finish it iteration.

A workaround for this was using a condition to check `Future.done()` at every `step()` call but that leads to a information delay.

Can this be done differently?

ygoumaz, 2023-06-26:
Are you using an asynchronous client call?

See https://docs.ros.org/en/humble/How-To-Guides/Sync-Vs-Async.html for more information. A solution using `future.done()` is also mentioned. I don't know any cleaner way to do it. I think I add tested the Emitter plugin when you submitted your PR and it was working fine. The delay shouldn't be over 1 timestep, or is it?

angel-ayala, 2023-06-29:
The issue happen when call the `Emitter.send_string` C++ service inside the python plugin, and according how the driver works seems to be that way since the `rclpp::spin_some()` happens after the execution of the python `step()` method and needs to use the `future.done()` method to check for the service status.
However, this behavior only allow to call `Emitter.send_string` service in one step yes, and the next step no, why? I'm not sure. Maybe is related to the moment when `rclpp::spin_some()` and `rclpy.spin_once()` are called, requiring to be sync.

However it seems to be an issue to use C++ and Python ROS node in the same controller and how the Nodes spin call are synced. Should the ROS nodes be running in a separated thread just as the `ros2_supervisor` works?