cyberbotics/webots_ros2 #796: ROS2 Supervisor unable to connect to Webots instance

jamestkpoon, 2023-06-29:
## What I tried
1. Install latest release (R2023b) from deb
2. ```
   sudo apt-get install ros-{ROS_DISTRO}-webots-ros2
   ```
3. ```
   ros2 launch webots_ros2_universal_robot multirobot_launch.py
   ```

## What happens
The webots application launches, but the robot arm does not move. Partial log is as follows:
```
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /usr/local/webots.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [webots-1]: process started with pid [919714]
[INFO] [ros2_supervisor.py-2]: process started with pid [919716]
[INFO] [ros2-3]: process started with pid [919720]
[INFO] [driver-4]: process started with pid [919722]
[driver-4] Cannot connect to Webots instance, retrying for another 50 seconds...
[ros2_supervisor.py-2] Cannot connect to Webots instance, retrying for another 50 seconds...
```

## Additional info
I also found [this issue](https://stackoverflow.com/questions/69099797/cant-connect-controller-to-specific-instance-of-webots) about setting ```WEBOTS_PID``` but that doesn't seem relevant since I can't predict the PID prior to launch.

ygoumaz, 2023-06-29:
We released today version 2023.1.0 of the package that will be compatible with Webots R2023b. It will be available at the next sync. Until then you can install the package from sources (2023.1.0 tag).

jamestkpoon, 2023-06-30:
Can confirm that building from source, and adding ```/usr/local/webots/lib/controller``` to ```LD_LIBRARY_PATH``` fixed this issue. Thanks again.

ygoumaz, 2023-06-30:
Normally you shouldn't have to add the Webots lib controller to `LD_LIBRARY_PATH`. The package exports its own version of the controller library and it should be used automatically.

jamestkpoon, 2023-06-30:
Just some observations below from my digging earlier today, for posterity.

Without modifying ```LD_LIBRARY_PATH```:
```
[driver-2] /home/james/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver: error while loading shared libraries: libController.so: cannot open shared object file: No such file or directory
[ERROR] [driver-2]: process has died [pid 47312, exit code 127, cmd '/home/james/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_l1kap
```

Modifying ```LD_LIBRARY_PATH``` to include ```/home/james/webots_ws/install/webots_ros2_driver/lib/controller```:
```
[driver-2] Cannot find any instance of Webots, retrying for another 50 seconds...
[ERROR] [driver-2]: process has died [pid 54470, exit code 1, cmd '/home/james/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_escp0msv'].
```
Webots console:
```
INFO: 'my_robot' extern controller: Waiting for local or remote connection on port 1234 targeting robot named 'my_robot'.
INFO: 'my_robot' extern controller: connected.
INFO: 'my_robot' extern controller: refusing connection attempt from another extern controller.
INFO: 'my_robot' extern controller: disconnected, waiting for new connection.
```

Finally with ```/usr/local/webots/lib/controller``` the [example project](https://docs.ros.org/en/galactic/Tutorials/Advanced/Simulators/Webots.html) connects, with the bridge successfully passes data between ROS2 and Webots.

ygoumaz, 2023-06-30:
I found the origin of your issue and why you need to set the `LD_LIBRARY_PATH` variable manually. In the new version of the package we introduced a new `WebotsController` node to simplify the declaration of custom robot plugins in launch files. Basically you can replace the declaration of the driver node by a simpler and clearer `WebotsController` one.

The goal was to ensure backwards compatibility so that old launch files still work, but apparently this is not the case. I will work on a patch to fix this issue and add a deprecation message. For information, the official ROS 2 documentation will also soon be updated to reflect this new node usage.

ygoumaz, 2023-06-30:
You can fix your launch file the following way:
```python
# replace this
my_robot_driver = Node(
    package='webots_ros2_driver',
    executable='driver',
    output='screen',
    additional_env={'WEBOTS_CONTROLLER_URL': controller_url_prefix() + 'my_robot'},
    parameters=[
        {'robot_description': robot_description},
    ]
)

# by this
my_robot_driver = WebotsController(
    robot_name='my_robot',
    parameters=[
        {'robot_description': robot_description},
    ]
)
```

jamestkpoon, 2023-06-30:
I just tried the suggested changes to the launch file, however unfortunately the issues with ```LD_LIBRARY_PATH``` remain the same.

ygoumaz, 2023-06-30:
Are you able to run examples?
```
ros2 launch webots_ros2_mavic robot_launch.py
```

jamestkpoon, 2023-06-30:
Just tried epuck and mavic examples, both crashed in the same way:
james@james-P6I7G50BKABC:~$ ros2 launch webots_ros2_mavic robot_launch.py
[INFO] [launch]: All log files can be found below /home/james/.ros/log/2023-06-30-18-46-49-101981-james-P6I7G50BKABC-66152
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /usr/local/webots.
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'ros2_supervisor.py' not found on the libexec directory '/home/james/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver' 

and the Webots app only shows up for a moment.

ygoumaz, 2023-06-30:
This is very weird. What does your `/home/james/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver` folder actually contain. The logs say that the python `ros2_supervisor` is not there. But when building from sources, it should be copied to that path.

ygoumaz, 2023-06-30:
Do you run into any errors on the webots_ros2_driver package when using `colcon build`?

jamestkpoon, 2023-06-30:
```

ygoumaz, 2023-07-03:
Your error is most likely coming from a conflict in detected libraries. I suggest to remove all occurences of webots_ros2 files and do a clean installation. 

Could you please try to clone and install from sources the last commit? In #805, I implemented the deprecation message for older launch files as discussed.

closed 2023-07-03 by ygoumaz