cyberbotics/webots_ros2 #843: [ROS2 Humble] WebotsController namespace Error

Tubagc, 2023-08-17:
I am trying to add namespace to the nodes and topics using the webots_ros2_turtlebot package. My only modification is specifying the namespace parameter when declaring the WebotsController object in the robot_launch.py 

`
turtlebot_driver = WebotsController(
        namespace='namespace',
        robot_name='TurtleBot3Burger',
        parameters=[
            {'robot_description': robot_description_path,
             'use_sim_time': use_sim_time,
             'set_robot_state_publisher': True},
            ros2_control_params
        ],
        remappings=mappings,
        respawn=True
    )
`

When I run the command

`
ros2 launch webots_ros2_turtlebot robot_launch.py
`
I get this error :

[image]

Does WebotsController class support namespace parameter?

ralphieraccoon, 2023-09-03:
I was stuck on this same issue for a long time! The solution is you need to include the namespace in your `ros2_control_params` YAML file for the controller manager and each individual controller in the format `<namespace>/controller_manager` and `<namespace>/<controller_name>`. See the `webots_ros2_universal_robot` example, in particular the files `ros2_control_abb_config.yaml` and `ros2_control_config.yaml`.

omichel, 2023-09-04:
I am closing the issue for now. Feel free to re-open if needed.

closed 2023-09-04 by omichel