cyberbotics/webots_ros2 #859: Colcon build returns error: package directory 'webots_ros2_importer/urdf2webots/urdf2webots' does not exist

hedaniel7, 2023-10-23:
**Describe the Bug**
`colcon build` returns error: package directory 'webots_ros2_importer/urdf2webots/urdf2webots' does not exist

**Steps to Reproduce**
1. clone the repo
2.  execute `colcon build`

**Expected behavior**
successful build

**Affected Packages**
List of affected packages:
  - webots_ros2_importer

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - ROS Version: Galactic
 - Operating System: Ubuntu 20.04

**Additional context**
```
dan@SurfBoard3:~/Projects/webots_ros2$ colcon build
[2.184s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'webots_ros2_driver' is in: /home/dan/Projects/CrazyflieControllers/install/webots_ros2_driver
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding webots_ros2_driver

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> webots_ros2_importer
Starting >>> webots_ros2_msgs                       
--- stderr: webots_ros2_importer                                                                   
error: package directory 'webots_ros2_importer/urdf2webots/urdf2webots' does not exist
---
Failed   <<< webots_ros2_importer [1.31s, exited with code 1]
Aborted  <<< webots_ros2_msgs [8.17s]                                 

Summary: 0 packages finished [10.4s]
  1 package failed: webots_ros2_importer
  1 package aborted: webots_ros2_msgs
  1 package had stderr output: webots_ros2_importer
  10 packages not processed

```

BruceXing24, 2023-10-30:
same problem, then I delete this importer package, it works up to now.

bonnuite, 2023-11-16:
Since urdf2webots folder in webots_ros2_importer is empty, as it was reported recently, in the cloned folder, option --recurse-submodules needed to automatically initialize and update each submodule in the repository:
_git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2.git -b fix-humble-ur-example_

patham9, 2024-10-06:
Your magic fixed it. Unfortunately ROS2 is still pretty much unusable together with Webots without knowing this quirk plus tons of others, all of which are destined to break over time. Strange that there isn't a Docker file or VM file with everything installed (ROS2 with Webots, not Webots alone which is useless on its own). So every user is required to waste their time with repeating the same arbitrary installation procedure that resulted form tons of bad practices leading to results not being reproducible.

takahashi-e6, 2025-02-06:
Hi

I've written the installation command on the following page. If anyone can close this issue, please do so.
https://github.com/cyberbotics/webots_ros2/wiki/Getting-Started

closed 2025-02-10 by hedaniel7