cyberbotics/webots_ros2 #861: webots_ros2_control : marked ‘override’, but does not override

233Xiaowu, 2023-10-25:
**Describe the Bug**
colcon build returns error:error: ‘rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn webots_ros2_control::Ros2ControlSystem::on_init(const hardware_interface::HardwareInfo&)’ marked ‘override’, but does not override

**Steps to Reproduce**
colcon build 

**Expected behavior**
colcon build successfully

**Affected Packages**
List of affected packages:
 webots_ros2_control 

**System**
 - Webots Version: 2023a
 - ROS Version: foxy
 - Operating System : Ubuntu 20.04

**Additional context**
--- stderr: webots_ros2_control                              
In file included from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:15:
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:58:79: error: ‘rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn webots_ros2_control::Ros2ControlSystem::on_init(const hardware_interface::HardwareInfo&)’ marked ‘override’, but does not override
   58 |     rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn on_init(
      |                                                                               ^~~~~~~
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:60:79: error: ‘rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn webots_ros2_control::Ros2ControlSystem::on_activate(const rclcpp_lifecycle::State&)’ marked ‘override’, but does not override
   60 |     rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn on_activate(
      |                                                                               ^~~~~~~~~~~
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:62:79: error: ‘rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn webots_ros2_control::Ros2ControlSystem::on_deactivate(const rclcpp_lifecycle::State&)’ marked ‘override’, but does not override
   62 |     rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn on_deactivate(
      |                                                                               ^~~~~~~~~~~~~
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:67:37: error: ‘hardware_interface::return_type webots_ros2_control::Ros2ControlSystem::read(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
   67 |     hardware_interface::return_type read(const rclcpp::Time & /*time*/, const rclcpp::Duration & /*period*/) override;
      |                                     ^~~~
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:68:37: error: ‘hardware_interface::return_type webots_ros2_control::Ros2ControlSystem::write(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
   68 |     hardware_interface::return_type write(const rclcpp::Time & /*time*/, const rclcpp::Duration & /*period*/) override;
      |                                     ^~~~~
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp: In member function ‘rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn webots_ros2_control::Ros2ControlSystem::on_init(const hardware_interface::HardwareInfo&)’:
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:87:46: error: ‘on_init’ is not a member of ‘hardware_interface::SystemInterface’
   87 |     if (hardware_interface::SystemInterface::on_init(info) !=
      |                                              ^~~~~~~
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp: In member function ‘virtual void webots_ros2_control::Ros2Control::step()’:
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:48:61: error: no matching function for call to ‘controller_manager::ControllerManager::read(rclcpp::Time, const rclcpp::Duration&)’
   48 |       mControllerManager->read(mNode->get_clock()->now(), dt);
      |                                                             ^
In file included from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2Control.hpp:23,
                 from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:15:
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:130:8: note: candidate: ‘void controller_manager::ControllerManager::read()’
  130 |   void read();
      |        ^~~~
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:130:8: note:   candidate expects 0 arguments, 2 provided
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:50:63: error: no matching function for call to ‘controller_manager::ControllerManager::update(rclcpp::Time, const rclcpp::Duration&)’
   50 |       mControllerManager->update(mNode->get_clock()->now(), dt);
      |                                                               ^
In file included from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2Control.hpp:23,
                 from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:15:
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:133:37: note: candidate: ‘controller_interface::return_type controller_manager::ControllerManager::update()’
  133 |   controller_interface::return_type update();
      |                                     ^~~~~~
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:133:37: note:   candidate expects 0 arguments, 2 provided
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:53:62: error: no matching function for call to ‘controller_manager::ControllerManager::write(rclcpp::Time, const rclcpp::Duration&)’
   53 |       mControllerManager->write(mNode->get_clock()->now(), dt);
      |                                                              ^
In file included from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2Control.hpp:23,
                 from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:15:
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:136:8: note: candidate: ‘void controller_manager::ControllerManager::write()’
  136 |   void write();
      |        ^~~~~
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:136:8: note:   candidate expects 0 arguments, 2 provided
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp: In member function ‘virtual void webots_ros2_control::Ros2Control::init(webots_ros2_driver::WebotsNode*, std::unordered_map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >&)’:
/home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:91:84: error: no matching function for call to ‘hardware_interface::ResourceManager::import_component(std::remove_reference<std::unique_ptr<webots_ros2_control::Ros2ControlSystemInterface>&>::type, __gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type&)’
[... 6 lines cut ...]
  151 |   void import_component(std::unique_ptr<ActuatorInterface> actuator);
      |        ^~~~~~~~~~~~~~~~
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:151:8: note:   candidate expects 1 argument, 2 provided
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:169:8: note: candidate: ‘void hardware_interface::ResourceManager::import_component(std::unique_ptr<hardware_interface::SensorInterface>)’
  169 |   void import_component(std::unique_ptr<SensorInterface> sensor);
      |        ^~~~~~~~~~~~~~~~
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:169:8: note:   candidate expects 1 argument, 2 provided
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:187:8: note: candidate: ‘void hardware_interface::ResourceManager::import_component(std::unique_ptr<hardware_interface::SystemInterface>)’
  187 |   void import_component(std::unique_ptr<SystemInterface> system);
      |        ^~~~~~~~~~~~~~~~
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:187:8: note:   candidate expects 1 argument, 2 provided
[... cut ...]
                 from /home/xiaowu/ROS_TRY/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:15:
/opt/ros/foxy/include/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = webots_ros2_control::Ros2ControlSystem; B = webots_ros2_control::Ros2ControlSystemInterface]’:
/opt/ros/foxy/include/class_loader/meta_object.hpp:216:7:   required from here
/opt/ros/foxy/include/class_loader/meta_object.hpp:218:12: error: invalid new-expression of abstract class type ‘webots_ros2_control::Ros2ControlSystem’
  218 |     return new C;

omichel, 2023-10-25:
Please note that foxy is deprecated and not maintained any more. I would recommend you to upgrade to humble / Ubuntu 22.04 / Webots R2023b.

233Xiaowu, 2023-10-27:
> Please note that foxy is deprecated and not maintained any more. I would recommend you to upgrade to humble / Ubuntu 22.04 / Webots R2023b.
Thanks for your advice ，I have changed my ubuntu and ROS version and It worked successfully.

closed 2023-10-27 by 233Xiaowu