cyberbotics/webots_ros2 #893: ROS2 Humble with Ubuntu 22.04 can't communicate with the demos robots

ruca03, 2024-02-05:
**Describe the Bug**
I installed the webots from the source files and when i run:        `ros2 pkg xml webots_ros2 | grep version`
It returns:
  `<version>2023.1.1</version>`

I start the demo with `ros2 launch webots_ros2_universal_robot multirobot_launch.py`.
While both arms spawn and the conveyor starts to move with the soda can on top,  both arms remain frozen.

The only warnings the webots console gives is when the soda can touches the arm:
 `WARNING: Contact joints between materials 'default' and 'default' will only be created for the 10 deepest contact points instead of all the 12 contact points.`

For what i could understand the controllers and the robots arent communicating with each other. Do you guys know what is happening or can you guys tell me what I'm doing wrong?

The terminal used to launch the demo gives the following response:

```
[INFO] [launch]: All log files can be found below /home/rui/.ros/log/2024-02-05-16-59-59-481893-rui-REPLANT-5596
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /home/rui/.ros/webotsR2023b/webots.
[INFO] [webots-1]: process started with pid [5597]
[INFO] [ros2_supervisor.py-2]: process started with pid [5599]
[INFO] [webots_controller_UR5e-3]: process started with pid [5602]
[INFO] [webots_controller_abbirb4600-4]: process started with pid [5605]
[INFO] [ros2-5]: process started with pid [5607]
[webots_controller_abbirb4600-4] Cannot connect to Webots instance, retrying for another 50 seconds...
[webots_controller_UR5e-3] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[ros2_supervisor.py-2] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_abbirb4600-4] The Webots simulation world is not yet ready, pending until loading is done...
[webots_controller_UR5e-3] The Webots simulation world is not yet ready, pending until loading is done...
[ros2_supervisor.py-2] The Webots simulation world is not yet ready, pending until loading is done...
[webots_controller_abbirb4600-4] [INFO] [1707152406.792231304] [resource_manager]: Initialize hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1707152406.792328131] [resource_manager]: Successful initialization of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1707152406.792438849] [resource_manager]: 'configure' hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1707152406.792444473] [resource_manager]: Successful 'configure' of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1707152406.792453572] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1707152406.792456950] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1707152406.795363743] [abb.controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[webots_controller_abbirb4600-4] [INFO] [1707152406.795778526] [abb.abbirb4600]: Controller successfully connected to robot in Webots simulation.
[ros2_supervisor.py-2] [WARN] [1707152406.842193865] [Ros2Supervisor.Ros2Supervisor]: The URDF Spawner is deprecated. It will be replaced by a new PROTOSpawner in the future. See https://github.com/cyberbotics/webots_ros2/issues/818 for more information.
[ros2_supervisor.py-2] [INFO] [1707152407.077071230] [Ros2Supervisor.Ros2Supervisor]: Ros2Supervisor has imported the URDF robot named "UR5e".
[INFO] [ur5e_controller-6]: process started with pid [5670]
[INFO] [abb_controller-7]: process started with pid [5672]
[INFO] [spawner-8]: process started with pid [5674]
[INFO] [spawner-9]: process started with pid [5676]
[INFO] [spawner-10]: process started with pid [5678]
[INFO] [spawner-11]: process started with pid [5680]
[INFO] [ros2-5]: process has finished cleanly [pid 5607]
[abb_controller-7] [INFO] [1707152407.388614352] [abb.abb_controller]: Waiting for action server to be ready...
[ur5e_controller-6] [INFO] [1707152407.459413697] [ur5e.ur5e_controller]: Waiting for action server to be ready...
[webots_controller_UR5e-3] [INFO] [1707152408.380589273] [resource_manager]: Initialize hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1707152408.380711450] [resource_manager]: Successful initialization of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1707152408.380846431] [resource_manager]: 'configure' hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1707152408.380852592] [resource_manager]: Successful 'configure' of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1707152408.380863707] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1707152408.380868009] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1707152408.385146517] [ur5e.controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[webots_controller_UR5e-3] [INFO] [1707152408.385616855] [ur5e.UR5e]: Controller successfully connected to robot in Webots simulation.
[spawner-9] [INFO] [1707152409.311290442] [spawner_ur_joint_state_broadcaster]: Waiting for '/ur5e/controller_manager' services to be available
[spawner-8] [INFO] [1707152409.312604206] [spawner_ur_joint_trajectory_controller]: Waiting for '/ur5e/controller_manager' services to be available
[spawner-10] [INFO] [1707152409.329472709] [spawner_abb_joint_trajectory_controller]: Waiting for '/abb/controller_manager' services to be available
[spawner-11] [INFO] [1707152409.409835686] [spawner_abb_joint_state_broadcaster]: Waiting for '/abb/controller_manager' services to be available
[spawner-9] [INFO] [1707152411.321595095] [spawner_ur_joint_state_broadcaster]: Waiting for '/ur5e/controller_manager' services to be available
[spawner-8] [INFO] [1707152411.322464644] [spawner_ur_joint_trajectory_controller]: Waiting for '/ur5e/controller_manager' services to be available
[spawner-10] [INFO] [1707152411.340501070] [spawner_abb_joint_trajectory_controller]: Waiting for '/abb/controller_manager' services to be available
[spawner-11] [INFO] [1707152411.420263965] [spawner_abb_joint_state_broadcaster]: Waiting for '/abb/controller_manager' services to be available
[spawner-9] [INFO] [1707152413.332441699] [spawner_ur_joint_state_broadcaster]: Waiting for '/ur5e/controller_manager' services to be available
[spawner-8] [INFO] [1707152413.332775907] [spawner_ur_joint_trajectory_controller]: Waiting for '/ur5e/controller_manager' services to be available
[spawner-10] [INFO] [1707152413.352002663] [spawner_abb_joint_trajectory_controller]: Waiting for '/abb/controller_manager' services to be available
[spawner-11] [INFO] [1707152413.432669351] [spawner_abb_joint_state_broadcaster]: Waiting for '/abb/controller_manager' services to be available
[spawner-8] [INFO] [1707152415.344530215] [spawner_ur_joint_trajectory_controller]: Waiting for '/ur5e/controller_manager' services to be available
[... 33 lines cut ...]
**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected Packages**
List of affected packages:
  - webots_ros2_A
  - webots_ros2_B

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Webots Version: [e.g., R2019b, R2020a revision 1]
 - ROS Version: [e.g., Dashing, Eloquent]
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

ralphieraccoon, 2024-02-06:
I've had the same problem with `iron`. I think `ros2_control` has made a breaking change recently.

Acwok, 2024-02-07:
Same problem with the e_puck demo:
```
[spawner-7] [INFO] [1707298270.095655670] [spawner_diffdrive_controller]: Waiting for '/controller_manager' services to be available
[spawner-8] [INFO] [1707298270.106261863] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
```

kotzw, 2024-02-08:
Sane probem on Ubuntu 22.04 on wsl2, humble.

wilhem, 2024-02-08:
Same problem here:
- Humble
- Ubuntu 22.04

omichel, 2024-02-08:
Can you try the patch proposed in #894 and let me know if that helps?

ruca03, 2024-02-08:
Hi, @omichel.

I did your sugegstion. While it stops the error message  "Waiting for ... to be available" the robots dont move.

The Webots log shows:
`INFO: 'abbirb4600' extern controller: Waiting for local or remote connection on port 1234 targeting robot named 'abbirb4600'.
INFO: conveyor_belt: Starting controller: /home/rui/.ros/webotsR2023b/webots/projects/objects/factory/conveyors/controllers/conveyor_belt/conveyor_belt -0.17 0
INFO: 'Ros2Supervisor' extern controller: Waiting for local or remote connection on port 1234 targeting robot named 'Ros2Supervisor'.
INFO: 'abbirb4600' extern controller: connected.
INFO: 'Ros2Supervisor' extern controller: connected.
INFO: 'UR5e' extern controller: Waiting for local or remote connection on port 1234 targeting robot named 'UR5e'.
INFO: 'UR5e' extern controller: connected.
WARNING: Contact joints between materials 'default' and 'default' will only be created for the 10 deepest contact points instead of all the 90 contact points.
WARNING: Contact joints between materials 'default' and 'default' will only be created for the 10 deepest contact points instead of all the 21 contact points.`

And the terminal when i run `ros2 launch webots_ros2_universal_robot multirobot_launch.py` shows:
```

[INFO] [launch]: All log files can be found below /home/rui/.ros/log/2024-02-08-15-43-50-895216-rui-REPLANT-18004
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: No valid Webots directory specified in `ROS2_WEBOTS_HOME` and `WEBOTS_HOME`, fallback to default installation folder /home/rui/.ros/webotsR2023b/webots.
[INFO] [webots-1]: process started with pid [18005]
[INFO] [ros2_supervisor.py-2]: process started with pid [18007]
[INFO] [webots_controller_UR5e-3]: process started with pid [18010]
[INFO] [webots_controller_abbirb4600-4]: process started with pid [18013]
[INFO] [ros2-5]: process started with pid [18015]
[webots_controller_UR5e-3] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_abbirb4600-4] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[ros2_supervisor.py-2] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_UR5e-3] The Webots simulation world is not yet ready, pending until loading is done...
[webots_controller_abbirb4600-4] The Webots simulation world is not yet ready, pending until loading is done...
[ros2_supervisor.py-2] The Webots simulation world is not yet ready, pending until loading is done...
[webots_controller_abbirb4600-4] [INFO] [1707407038.230590805] [resource_manager]: Initialize hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1707407038.230821465] [resource_manager]: Successful initialization of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1707407038.231043607] [resource_manager]: 'configure' hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1707407038.231060032] [resource_manager]: Successful 'configure' of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1707407038.231074364] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1707407038.231079585] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1707407038.235500788] [abb.controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[webots_controller_abbirb4600-4] [INFO] [1707407038.236073859] [abb.abbirb4600]: Controller successfully connected to robot in Webots simulation.
[ros2_supervisor.py-2] [WARN] [1707407038.415912712] [Ros2Supervisor.Ros2Supervisor]: The URDF Spawner is deprecated. It will be replaced by a new PROTOSpawner in the future. See https://github.com/cyberbotics/webots_ros2/issues/818 for more information.
[ros2_supervisor.py-2] [INFO] [1707407038.694156319] [Ros2Supervisor.Ros2Supervisor]: Ros2Supervisor has imported the URDF robot named "UR5e".
[INFO] [ur5e_controller-6]: process started with pid [18086]
[INFO] [abb_controller-7]: process started with pid [18088]
[INFO] [ros2_control_node-8]: process started with pid [18090]
[INFO] [ros2_control_node-9]: process started with pid [18092]
[INFO] [ros2_control_node-10]: process started with pid [18094]
[INFO] [ros2_control_node-11]: process started with pid [18096]
[ros2_control_node-9] [WARN] [1707407038.727792944] [controller_manager]: 'update_rate' parameter not set, using default value.
[ros2_control_node-9] [INFO] [1707407038.727924904] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-9] [INFO] [1707407038.728784460] [controller_manager]: update rate is 100 Hz
[ros2_control_node-9] [INFO] [1707407038.728917604] [controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-11] [WARN] [1707407038.733139987] [controller_manager]: 'update_rate' parameter not set, using default value.
[ros2_control_node-11] [INFO] [1707407038.733226851] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-8] [WARN] [1707407038.733706203] [controller_manager]: 'update_rate' parameter not set, using default value.
[... cut ...]
[webots_controller_UR5e-3] [INFO] [1707407039.703442747] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1707407039.703447273] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1707407039.708499371] [ur5e.controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[webots_controller_UR5e-3] [INFO] [1707407039.709177077] [ur5e.UR5e]: Controller successfully connected to robot in Webots simulation.

```

ralphieraccoon, 2024-02-09:
I'm wondering if ros-controls/ros2_control#1169 might be the culprit here.

ralphieraccoon, 2024-02-09:
I'm wondering if ros-controls/ros2_control#1169 might be the culprit. It says it is API breaking.

elad-eyal, 2024-02-09:
Fails for me too, on Ubuntu 22.04 , WSL.

I installed using `apt` the packages `ros-humble-ros2-control` and `ros-humble-ros2-controllers` so I get ros2_control v2.37.0 (That's what I get when trying `ros2 pkg xml ros2_control`

elad-eyal, 2024-02-14:
I wish there was a fix or workaround @omichel  .. Webots is so nice but I need to use ros2_control

ralphieraccoon, 2024-02-14:
Based on my analysis, the problem is the `robot_description` topic has no publishers. The `controller_manager` services that are being polled for are not made available until a message is published on this topic. So I think something is failing which prevents the `robot-description` publisher from being created.

I've also tried loading the URDF directly into the resource manager, which I believe bypasses the need to subscribe to `robot_description`, but I've not got it work without error.

skpawar1305, 2024-02-14:
> I wish there was a fix or workaround @omichel .. Webots is so nice but I need to use ros2_control

you may use build ros2_control 2.36.1 for now

closed 2024-02-15 by omichel

wilhem, 2024-02-15:
How to update our systems with the new changes?
Doing a:
```
$sudo apt update 
$sudo apt upgrade 
```

does show, that the repository to webots_ros2 is checked but no update is downloaded.

wilhem, 2024-03-02:
And there is any estimated time for the next release?

omichel, 2024-03-02:
Now that all the tests are passing on all ROS2 distributions, @Yadunund may be able to bump a release. A PR for this was prepared a long time ago in #839 and @Yadunund also prepared one in #892. Not sure from which we should re-start.

Yadunund, 2024-03-04:
If CI is working on all distro we should continue with #839 imo

omichel, 2024-03-04:
> If CI is working on all distro we should continue with #839 imo

I agree. Can you take the lead on this since @ygoumaz is not working on the project any more?

wilhem, 2024-04-07:
This issue is still continuing here: [link](https://github.com/cyberbotics/webots_ros2/issues/910)