cyberbotics/webots_ros2 #895: Explanation about the parameter: 'robot_description'

wilhem, 2024-02-11:
**Documentation page(s) concerned**
[Link to the concerned page.](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/launch/robot_launch.py)

**What is the problem with the current documentation?**
In many launch files, the robot_description parameter has been replace by the following string:
'<robot name=""><link name=""/></robot>'

**Describe how you would fix it**
What's the purpose of it? Why it cannot be passed the whole urdf/xacro content to that parameter?

wilhem, 2024-02-14:
Yes, I see.

The code I posted does not get formatted in this interface.
So I post a link to that line here: [link](https://github.com/cyberbotics/webots_ros2/blob/1812230265f654b87296993c37894ff21eb2530a/webots_ros2_turtlebot/launch/robot_launch.py#L53)

skpawar1305, 2024-02-15:
> **Documentation page(s) concerned** [Link to the concerned page.](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/launch/robot_launch.py)
> 
> **What is the problem with the current documentation?** In many launch files, the robot_description parameter has been replace by the following string: [Link](https://github.com/cyberbotics/webots_ros2/blob/1812230265f654b87296993c37894ff21eb2530a/webots_ros2_turtlebot/launch/robot_launch.py#L53)
> 
> **Describe how you would fix it** What's the purpose of it? Why it cannot be passed the whole urdf/xacro content to that parameter?

it's just dummy, if set_robot_state_publisher param of the WebotsController is set to true, the mentioned data gets overriden with a service call
(I can be wrong though)