cyberbotics/webots_ros2 #910: Unable to use all examples

LiuXianBang, 2024-03-12:
**Describe the Bug**
Unable to use all examples to slam or nav,including TurtleBot3 E-puck and so on. 

here is the node graph
[image]

noticed the topic /scan and /imu didn't been subscribe yet,but i already use ros2 launch webots_ros2_turtlebot robot_launch.py slam:=true

here is the console log
[image]
maybe the wrong in here but i cant find the way to fix this
[image]
 
my english is poor,hope to receive your help

wilhem, 2024-03-18:
It is correlated to this bug [here](https://github.com/cyberbotics/webots_ros2/issues/893).
Despite the fact, that Cyberbotics found allegedly a fix for that bug (which is caused by ros control package), the fix is still not be available yet.
In other words: we need to hope that Cyberbotics release an update soon.

LiuXianBang, 2024-03-22:
thank for comment，but this fix still not working for me :cry

wilhem, 2024-03-22:
Because the package has been updated yet.
Still waiting for the new release....

YoungRicky7, 2024-04-07:
Is it fixed now?

wilhem, 2024-04-07:
Sadly not. Or better. the bug has been fixed, but @Yadunund or Cyberbotics in general didn't release the change though the repository.
Hopefully it will come soon

LiuXianBang, 2024-04-07:
Maybe they forgot they haven't released the new version yet，anyway to communicate  with them easily？

YoungRicky7, 2024-04-07:
> Maybe they forgot they haven't released the new version yet，anyway to communicate with them easily？

Nah, as for me I give up using WEBOTS, I use ignition gazebo as the substitute.

skpawar1305, 2024-04-07:
[ros-humble-webots-ros2-control_2023.1.1-0jammy_amd64.zip](https://github.com/cyberbotics/webots_ros2/files/14896695/ros-humble-webots-ros2-control_2023.1.1-0jammy_amd64.zip)

you can extract n install this
I've compiled it n there's no fishy thing in it 👻

wilhem, 2024-04-07:
Thanks @skpawar1305 but there are some problems because of the fact, that you are not part of cyberbotics.
Not everyone here is just a private person. Some people need an official version from cyberbotics.
By the way, your packages is for jammy only. But many of us use still "humble"

skpawar1305, 2024-04-07:
> Thanks @skpawar1305 but there are some problems because of the fact, that you are not part of cyberbotics. Not everyone here is just a private person. Some people need an official version from cyberbotics. By the way, your packages is for jammy only. But many of us use still "humble"

yes, I had shared it with a student who had build errors. n it's for humble

omichel, 2024-04-08:
I personally don't have the bandwidth to make this release now, but I am happy to help any volunteer to proceed with the release. The release procedure is explained [here](https://github.com/cyberbotics/webots_ros2/wiki/Internal-Publishing) and a PR for preparing the release is already open in #839. Just let me know.

wilhem, 2024-04-08:
@skpawar1305 can you prepare a release for everybody based on your package?

skpawar1305, 2024-04-08:
> @skpawar1305 can you prepare a release for everybody based on your package?

not this week, next week I can 👍

omichel, 2024-04-08:
I found some time this afternoon to proceed with the release. So, now we need to wait for the approval from the ROS 2 maintainers of these 3 PRs:
- [rolling](https://github.com/ros/rosdistro/pull/40507)
- [humble](https://github.com/ros/rosdistro/pull/40509)
- [iron](https://github.com/ros/rosdistro/pull/40510)
Then, the corresponding packages will be published at the next sync.

omichel, 2024-04-09:
**Update**: all the 3 PRs were merged. We are now waiting for the next sync which should happen within a couple of weeks or so on the different distros:

- [rolling](http://repo.ros2.org/status_page/ros_rolling_default.html?q=SYNC)
- [humble](http://repo.ros2.org/status_page/ros_humble_default.html?q=SYNC)
- [iron](http://repo.ros2.org/status_page/ros_iron_default.html?q=SYNC)

VincentDelort, 2024-05-07:
Hi @omichel , when do you believe the fixes will be merged into public (`apt`) version of ros humble (I do not know how to interpret the pages that you linked)?

omichel, 2024-05-07:
It should be already released now. See https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2024-04-18/37272

closed 2024-05-07 by omichel