cyberbotics/webots_ros2 #917: The robot is already connected or your IP address is not allowed by this instance of Webots.

flyingdoraemon, 2024-04-02:
**Describe the Bug**
After follow the tutorial in [Installation (Windows)](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html) and try the [Setting up a robot simulation (Basic)](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Webots/Setting-Up-Simulation-Webots-Basic.html#setting-up-a-robot-simulation-basic), after executing `ros2 launch my_package robot_launch.py`, the problem is up above.

**Steps to Reproduce**
1. Installed ROS2 + ros-iron-webots-ros2 on WSL2, Ubuntu-22.04
2. Installed Webots on Windows11
3. Configured workspace, sourced ROS, with WEBOTS_HOME=/mnt/d/Programs Files/Webots
4. Started ros2 launch my_package robot_launch.py

**Screenshots**
```
$ ros2 launch my_package robot_launch.py
[INFO] [launch]: All log files can be found below /home/...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [6539]
[INFO] [webots_controller_my_robot-2]: process started with pid [6541]
[webots_controller_my_robot-2] The Webots simulation world is not yet ready, pending until loading is done...
[webots_controller_my_robot-2] Error: The connection was closed by Webots. The robot is already connected or your IP address is not allowed by this instance of Webots.
[webots_controller_my_robot-2] [ros2run]: Process exited with failure 1
[ERROR] [webots_controller_my_robot-2]: process has died [pid 6541, exit code 1, cmd '/home/.../ros2_ws/install/webots_ros2_driver/share/webots_ros2_driver/scripts/webots-controller --robot-name=my_robot --protocol=tcp --ip-address=172.24.128.1 --port=1234 ros2 --ros-args -p robot_description:=/home/.../ros2_ws/install/my_package/share/my_package/resource/my_robot.urdf'].
[INFO] [webots-1]: process has finished cleanly [pid 6539]
```

**System**
 - Webots Version: R2023b
 - ROS Version: `$ ros2 pkg xml webots_ros2 | grep version` : ` <version>2023.1.1</version>`
 - WSL Version: 2.1.5.0
 - Operating System:  Window11

**Additional context**
The previous problem is [here](https://robotics.stackexchange.com/questions/110345/extern-controller-cannot-connect-to-webots-instance), and the question in [this comment](https://github.com/cyberbotics/webots_ros2/issues/759#issuecomment-1562996470) is the same, in #759 .

As is mentioned in #879 , but when I tried to add `.wslconfig`, the error is `[webots_controller_my_robot-2] Cannot connect to Webots instance, retrying for another 50 seconds...`
After deleting `.wslconfig`, the error is mentioned above.

 The method in [this comment](https://github.com/cyberbotics/webots_ros2/issues/759#issuecomment-1566072315) helped me checked there are no port 1234 in used and there are no firewall issues. And the test: `telnet 172.24.128.1 1234` while webots is running is ok.

The output of running demo packages is the same, like `ros2 launch webots_ros2_universal_robot multirobot_launch.py`, except that it will endlessly try to reconnect.

I tried to search for methods but failed, please help and thanks a lot!!!

ZJZ0405, 2024-07-05:
I try to run “/opt/ros/humble/share/webots_ros2_driver/scripts/webots-controller --robot-name=my_robot --protocol=tcp --ip-address=10.255.255.254 --port=1234 ros2 --ros-args -p robot_description:=/home/mol/ws/my_package/install/my_package/share/my_package/resource/my_robot.urdf”, and change "--ip-address=10.255.255.254". It works.

andrespineda, 2025-02-16:
I spent over a full day trying to figure this out and found a solution.

In my case, the issue was that the program located in "webots_ros2_driver/lib/python3.12/site-packages/webots-driver/utils.py" has a function named "get_wsl_ip_address()" . This function gets the "nameserver" ip address from the "/etc/resolv.conf" file and returns it incorrectly as the ip address of the Windows host. Because of this, the Ubuntu webots controller keeps trying to connect to the webots simulator running on the Windows side using the nameserver ip address as the TCP IP address.  I my system, the nameserver was set to 10.255.255.  The robot_launch.py kept trying to connect to TCP 10.255.255.255 port 1234. 

To prove that this was the problem, I manually edited the "/etc/resolv.conf" file and changed the nameserver address to "127.0.0.1".

This worked:  I was then able to run "ros2 launch webots_ros2_epuck robot_launch.py"" and it connected up to the webot simulator.

Changing the nameserver address is probably not the correct way to fix the problem, but it got me to a working demo and can move on.

I consider the code in utils.py to be an error as the nameserver ip address is not the correct value to use to connect to the Windows host webot simulator.