cyberbotics/webots_ros2 #988: [ERROR] [controller_manager]: Could not initialize the controller named 'diffdrive_controller'

SpaceSoarer90, 2024-09-30:
**Describe the Bug**
[webots_controller_TurtleBot3Burger-5] [ERROR] [1727690103.628136189] [controller_manager]: Could not initialize the controller named 'diffdrive_controller'
[webots_controller_TurtleBot3Burger-5] Exception thrown during init stage with message: Invalid value set during initialization for parameter 'left_wheel_names': Parameter 'left_wheel_names' cannot be empty.

Trying the webots_ros2_turtlebot package but fails on the 'diffdrive_controller' node. This makes me unable to use the teleop nodes. I know that it uses "/opt/ros/jazzy/share/webots_ros2_turtlebot/resource/ros2control.yml" as the --params-file, however, that seems to be complete.

```
...
diffdrive_controller:
  ros__parameters:
    left_wheel_names: ["left wheel motor"]
    right_wheel_names: ["right wheel motor"]
...
```

**Steps to Reproduce**
1. Install necessary packages for webots_ros2_turtlebot (webots, ros2, ros-jazzy-ros2-turtle
2. `$ ros2 launch webots_ros2_turtlebot robot_launch.py`
3. See terminal for the error.

**Expected behavior**
Being able to drive the robot with teleop nodes.

**Affected Packages**
List of affected packages:
  - webots_ros2_turtlebot

**Screenshots**
[image]
[image]

**System**
 - Webots Version: [e.g., R2019b, R2020a revision 1]
 - ROS Version: [e.g., Dashing, Eloquent]
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
The wifi card of my laptop is not working, maybe that is the problem? The webots_controller python script uses ipc by default but maybe using tcp would be better?

patham9, 2024-10-25:
Webots has been broken for quite some time for ROS2, I suggest using Gazebo.

legubiao, 2024-11-16:
Hi I also enter this problem, and I solved it by modifying the launch file `spawner` parts to add the param file, like this:
```py
    diff_drive_controller_spawner = Node(
        package='controller_manager',
        executable='spawner',
        arguments=["diff_drive_controller",
                   "--param-file", ros2_control_params] + controller_manager_timeout,
    )
```

I tested on ubuntu 24.04, ROS2 Jazzy, Webots 2023A from apt install.

wentywenty, 2025-04-14:
I reran the turtlebot example following your scenario and it still loads incorrectly, here is the log

```
duzhong@ucar-mini:~/sha32$ ros2 launch sha32_gazebo webot.launch.py
[INFO] [launch]: All log files can be found below /home/duzhong/.ros/log/2025-04-15-01-04-23-346120-ucar-mini-153630
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots_controller_TurtleBot3Burger-1]: process started with pid [153631]
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650264.859338823] [resource_manager]: Initialize hardware 'WebotsControl' 
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650264.859498430] [resource_manager]: Successful initialization of hardware 'WebotsControl'
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650264.859537265] [resource_manager]: 'configure' hardware 'WebotsControl' 
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650264.859546538] [resource_manager]: Successful 'configure' of hardware 'WebotsControl'
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650264.859550818] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650264.859554352] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650265.426325943] [plugin_node_example]:   - properties: {'parameterExample': 'someValue'}
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650265.426773469] [plugin_node_example]:   - robot name: TurtleBot3Burger
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650265.427079465] [plugin_node_example]:   - basic timestep: 20
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650265.427346595] [plugin_node_example]:   - is supervisor? False
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650265.429660924] [TurtleBot3Burger]: Controller successfully connected to robot in Webots simulation.
[INFO] [spawner-2]: process started with pid [153682]
[INFO] [spawner-3]: process started with pid [153684]
[spawner-2] [INFO] [1744650266.383997651] [spawner_diffdrive_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-2] [INFO] [1744650266.641769252] [spawner_diffdrive_controller]: Setting controller param "params_file" to "['/home/duzhong/sha32/ros2control.yml']" for diffdrive_controller
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650266.645996072] [controller_manager]: Loading controller 'diffdrive_controller'
[spawner-2] [INFO] [1744650266.659762842] [spawner_diffdrive_controller]: Loaded diffdrive_controller
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650266.661621890] [controller_manager]: Configuring controller 'diffdrive_controller'
[webots_controller_TurtleBot3Burger-1] [ERROR] [1744650271.669234641] [controller_manager]: Switch controller timed out after 5.000000 seconds!
[spawner-2] [ERROR] [1744650271.670326125] [spawner_diffdrive_controller]: Failed to activate controller : diffdrive_controller
[spawner-3] [INFO] [1744650271.692668478] [spawner_joint_state_broadcaster]: Setting controller param "params_file" to "['/home/duzhong/sha32/ros2control.yml']" for joint_state_broadcaster
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650271.699331369] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-3] [INFO] [1744650271.719464441] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650271.721544062] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[webots_controller_TurtleBot3Burger-1] [INFO] [1744650271.721729103] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ERROR] [spawner-2]: process has died [pid 153682, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diffdrive_controller --param-file /home/duzhong/sha32/ros2control.yml --controller-manager-timeout 50 --ros-args'].
[webots_controller_TurtleBot3Burger-1] [ERROR] [1744650276.729026284] [controller_manager]: Switch controller timed out after 5.000000 seconds!
[spawner-3] [ERROR] [1744650276.729990167] [spawner_joint_state_broadcaster]: Failed to activate controller : joint_state_broadcaster
[ERROR] [spawner-3]: process has died [pid 153684, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --param-file /home/duzhong/sha32/ros2control.yml --controller-manager-timeout 50 --ros-args'].
```

> 你好，我也遇到了这个问题，我通过修改启动文件`spawner`部分来添加参数文件来解决它，如下所示：
> 
>     diff_drive_controller_spawner = Node(
>         package='controller_manager',
>         executable='spawner',
>         arguments=["diff_drive_controller",
>                    "--param-file", ros2_control_params] + controller_manager_timeout,
>     )
> 我通过 apt install 在 ubuntu 24.04、ROS2 Jazzy、Webots 2023A 上进行了测试。