cyberbotics/webots_ros2 #1015: Cannot launch webots_ros2_universal_robot after updated ubuntu22.04

lkx8421, 2024-12-18:
when I used 'sudo apt updated' on ubuntu 22.04, I cannot use webots ros2 example 'ros2 launch webots_ros2_universal_robot multirobot_launch.py', log is 

[INFO] [webots-1]: process started with pid [113805]
[INFO] [ros2_supervisor.py-2]: process started with pid [113807]
[INFO] [webots_controller_UR5e-3]: process started with pid [113810]
[INFO] [webots_controller_abbirb4600-4]: process started with pid [113813]
[INFO] [ros2-5]: process started with pid [113815]
[webots_controller_UR5e-3] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_abbirb4600-4] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[ros2_supervisor.py-2] The specified robot is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_UR5e-3] The specified robot is not in the list of robots with <extern> controllers, retrying for another 45 seconds...
[webots_controller_abbirb4600-4] Warning: Webots [R2023b] and libController [R2024a] versions are not the same for Robot 'abbirb4600'! Different versions can lead to undefined behavior.
[ros2_supervisor.py-2] Warning: Webots [R2023b] and libController [R2024a] versions are not the same for Robot 'Ros2Supervisor'! Different versions can lead to undefined behavior.
[webots_controller_abbirb4600-4] [INFO] [1734507005.729230951] [resource_manager]: Initialize hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1734507005.729269366] [resource_manager]: Successful initialization of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1734507005.729296123] [resource_manager]: 'configure' hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1734507005.729299426] [resource_manager]: Successful 'configure' of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] [INFO] [1734507005.729305899] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_abbirb4600-4] [INFO] [1734507005.729308036] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_abbirb4600-4] terminate called after throwing an instance of 'std::runtime_error'
[webots_controller_abbirb4600-4]   what():  Hardware name WebotsControl is duplicated. Please provide a unique 'name' in the URDF.
[webots_controller_abbirb4600-4] [ros2run]: Aborted
[ERROR] [webots_controller_abbirb4600-4]: process has died [pid 113813, exit code 250, cmd '/opt/ros/humble/share/webots_ros2_driver/scripts/webots-controller --robot-name=abbirb4600 --protocol=ipc --port=1234 ros2 --ros-args -r __ns:=/abb -p robot_description:=/opt/ros/humble/share/webots_ros2_universal_robot/resource/webots_abb_description.urdf -p use_sim_time:=True -p set_robot_state_publisher:=True --params-file /opt/ros/humble/share/webots_ros2_universal_robot/resource/ros2_control_abb_config.yaml'].

Heeeeena, 2024-12-19:
I encountered the same problem. Previously, some of my feature packs and examples in Webots_ros2 were working properly, but for some reason, during recent runtime, all the examples are no longer working properly. I am using Webots2023b+Ubuntu 22.04 (WSL2)+Windows 11. When I start the launch file, the controller starts to report errors and disconnect from Webots2023 after connecting to it. Due to respawn=true, this process will continue to repeat and the connection will not be able to connect.
[image]
[image]

Heeeeena, 2024-12-19:
> 我也遇到了同样的问题，之前我的Webots_ros2中的一些功能包和示例都正常工作，但不知何故，在最近的运行过程中，所有示例都不再正常工作。我使用的是Webots2023b+Ubuntu 22.04 (WSL2)+Windows 11。当我启动启动文件时，控制器开始报错，并在连接到Webots2023后断开连接。由于respawn=true，这个过程会不断重复，连接将无法连接。 ![图像](https://private-user-images.githubusercontent.com/113170953/397174994-63a68655-edc7-40b2-8abd-9b7e6799af8d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1NzEzNTYsIm5iZiI6MTczNDU3MTA1NiwicGF0aCI6Ii8xMTMxNzA5NTMvMzk3MTc0OTk0LTYzYTY4NjU1LWVkYzctNDBiMi04YWJkLTliN2U2Nzk5YWY4ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxOVQwMTE3MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yOTJjMzgzYWI3NjE4MWYxODBiOGIwZTc0ZmMwMGRkYTQ1M2Y0YjAwMmE1ZjJkMjYxYTFmZGQ2ZmIxYTU2MmM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.vT19IUNII75XYcn450OIpX1-BPbsJb-c14hnOjPrv-4) ![图像](https://private-user-images.githubusercontent.com/113170953/397175718-0764a71c-4aad-43e2-bf04-5aea5c163e37.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1NzEzNTYsIm5iZiI6MTczNDU3MTA1NiwicGF0aCI6Ii8xMTMxNzA5NTMvMzk3MTc1NzE4LTA3NjRhNzFjLTRhYWQtNDNlMi1iZjA0LTVhZWE1YzE2M2UzNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxOVQwMTE3MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNTkyOGFkZWZiMDFkNDMyZGZlOTkxMjk4M2ZjMzEwODg3MzllMjI1YWRkN2I2YjU5Yjc2ZTVmZWQ0ZTMxZWZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xmNkvn7Q-R4unEFcvduqP5one-M_7xu0OZBBCIWuSwg)

I noticed that someone had already mentioned this issue #1011 earlier, and when I followed their method to change the relevant content in the webots_ros2_control package and recompiled it, the problem was solved. Thanks very much!

2625102155, 2025-01-13:
> I noticed that someone had already mentioned this issue #1011 earlier, and when I followed their method to change the relevant content in the webots_ros2_control package and recompiled it, the [...]

I changed it according to their method, but there is still this problem, can you please explain in detail how you modified it, thank you

closed 2025-01-14 by lkx8421