cyberbotics/webots_ros2 #1031: Webots Ros2 program (any of the demos) running in Ubuntu (Windows WSL2) is not able to connect to Webots host simulator in Windows 11.

andrespineda, 2025-02-16:
**Describe the Bug**
After installing Webots on a Windows 11 host configured with a Ubuntu 22.04 on WSL2, any of the demo programs fail in trying to connect to the Windows Webot simulator. 

**Steps to Reproduce**
1. Install the Ros2 and Webots code as specified in the Ros2 documentation: https://docs.ros.org/en/foxy/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html#install-wsl2https://docs.ros.org/en/foxy/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html#install-wsl2.
2. After installation, attempt to run the provided demo script:  ros2 launch webots_ros2_universal_robot multirobot_launch.py.

**Expected behavior**
The expectation is that the code will start to run and pop open the Webots simulator window on the Windows host. It will them start to send commands to the robot running in the simulation to perform the demo actions.
Instead, the program starts up, but is unable to connect to the host. The screen continuously displays an error that it cannot connect and will try in a few seconds. It never is able to connect.
 
**Affected Packages**
List of affected packages:
 All of the demo programs, such as epuck, universal_robot, etc.

**Screenshots**
Not available

**System**
 - Webots Version: [R2025a ]
 - ROS Version: [Jazzy]
 - Operating System: [Windows 11, WSL2 Ubuntu 24.04.2 LTS]
 - Graphics Card: [Unknown]

**Additional context**

I spent over a full day trying to figure this out and found a solution.

In my case, the issue was that the program located in "webots_ros2_driver/lib/python3.12/site-packages/webots-driver/utils.py" has a function named "get_wsl_ip_address()" . This function gets the "nameserver" ip address from the WSL2  "/etc/resolv.conf" file and returns it incorrectly as the ip address of the Windows host. Because of this, the Ubuntu webots controller keeps trying to connect to the webots simulator running on the Windows side using the nameserver ip address as the TCP IP address. I my system, the nameserver was set to 127.0.0.53. The "webots-controller"  kept trying to connect to TCP 127.0.0.53 port 1234.

To prove that this was the problem, I manually edited the "/etc/resolv.conf" file and changed the nameserver address to "127.0.0.1".

This worked: I was then able to run "ros2 launch webots_ros2_epuck robot_launch.py"" and it connected up to the webot simulator using TCP 127.0.0.1 Port 1234.  Of course this only works while the terminal is open. Once I close the terminal and open it again, the /etc/resolv changes and it starts to fail again.

Changing the nameserver address is not the correct way to fix the problem, but it got me to a working demo and can move on.

I consider the code in utils.py to be an error as the nameserver ip address is not the correct value to use to connect to the Windows host webot simulator.

pantagnun, 2025-02-28:
Hi, pardon. I meet the same problem.I have tried the "Execute_Process" to choose the right windows ip I needed for webots_controller connection and it worked, but the problem occured then. It throwed the exception like below：
[webots-controller-2] terminate called after throwing an instance of 'std::runtime_error'
[webots-controller-2]   what():  Error: The Python module with the WebotsNode class cannot be executed.
[webots-controller-2] [ros2run]: Aborted
Have you met the problem too? could you tell me how to fix it? please thx

myudak, 2025-04-24:
is there any simpler solutions? i have the exact same issue