cyberbotics/webots_ros2 #1034: Cannot execute the joint_state_controller and the joint_state_broadcaster of the example multi-robot Universal Robot

samagalhaes, 2025-03-20:
I am trying to execute the universal robot multi-robot example, but I am having an error launching the joint_state_controller and the Joint_state_broadcaster. It is having this error:

```bash
[webots_controller_abbirb4600-4] [ERROR] [1742483730.402561926] [abb.controller_manager]: Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'abb_joint_trajectory_controller': failed to parse arguments: Couldn't parse parameter override rule: '-p -p'. Error: Expecting token or wildcard, at ./src/rcl/arguments.c:1325, at ./src/rcl/arguments.c:352
```

The full log is below

```bash
samagalhaes@samagalhaes-tribelab:~/ros_ws$ ros2 launch webots_ros2_universal_robot multirobot_launch.py 

[INFO] [launch]: All log files can be found below /home/samagalhaes/.ros/log/2025-03-20-15-15-21-177096-samagalhaes-tribelab-157601
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [157605]
[INFO] [ros2_supervisor.py-2]: process started with pid [157606]
[INFO] [webots_controller_UR5e-3]: process started with pid [157607]
[INFO] [webots_controller_abbirb4600-4]: process started with pid [157609]
[INFO] [ros2-5]: process started with pid [157611]
[webots_controller_abbirb4600-4] The specified robot (at /tmp/webots/samagalhaes/1234/ipc/abbirb4600/extern) is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_UR5e-3] The specified robot (at /tmp/webots/samagalhaes/1234/ipc/UR5e/extern) is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[ros2_supervisor.py-2] The specified robot (at /tmp/webots/samagalhaes/1234/ipc/Ros2Supervisor/extern) is not in the list of robots with <extern> controllers, retrying for another 50 seconds...
[webots_controller_UR5e-3] The specified robot (at /tmp/webots/samagalhaes/1234/ipc/UR5e/extern) is not in the list of robots with <extern> controllers, retrying for another 45 seconds...
[webots_controller_abbirb4600-4] [INFO] [1742483727.487086742] [abb.controller_manager]: Subscribing to '/abb/robot_description' topic for robot description.
[webots_controller_abbirb4600-4] [INFO] [1742483727.488251963] [abb.abbirb4600]: Controller successfully connected to robot in Webots simulation.
[ros2_supervisor.py-2] [WARN] [1742483727.534425574] [Ros2Supervisor.Ros2Supervisor]: The URDF Spawner is deprecated. It will be replaced by a new PROTOSpawner in the future. See https://github.com/cyberbotics/webots_ros2/issues/818 for more information.
[ros2_supervisor.py-2] [INFO] [1742483727.719146054] [Ros2Supervisor.Ros2Supervisor]: Ros2Supervisor has imported the URDF robot named "UR5e".
[INFO] [ur5e_controller-6]: process started with pid [157800]
[INFO] [abb_controller-7]: process started with pid [157801]
[INFO] [robot_state_publisher-8]: process started with pid [157802]
[INFO] [robot_state_publisher-9]: process started with pid [157803]
[INFO] [spawner-10]: process started with pid [157804]
[INFO] [spawner-11]: process started with pid [157805]
[INFO] [spawner-12]: process started with pid [157806]
[INFO] [spawner-13]: process started with pid [157807]
[INFO] [ros2-5]: process has finished cleanly [pid 157611]
[robot_state_publisher-9] [INFO] [1742483728.102487824] [ur5e.robot_state_publisher]: Robot initialized
[robot_state_publisher-8] [INFO] [1742483728.107465252] [abb.robot_state_publisher]: Robot initialized
[webots_controller_abbirb4600-4] [INFO] [1742483728.107536573] [abb.controller_manager]: Received robot description from topic.
[webots_controller_abbirb4600-4] terminate called after throwing an instance of 'std::runtime_error'
[webots_controller_abbirb4600-4]   what():  URDF cannot be parsed by a `ros2_control` component parser: no ros2_control tag
[spawner-11] [INFO] [1742483728.136447433] [spawner_ur_joint_state_broadcaster]: waiting for service /ur5e/controller_manager/list_controllers to become available...
[spawner-10] [INFO] [1742483728.136790457] [spawner_ur_joint_trajectory_controller]: waiting for service /ur5e/controller_manager/list_controllers to become available...
[spawner-12] [INFO] [1742483728.145509691] [spawner_abb_joint_trajectory_controller]: waiting for service /abb/controller_manager/list_controllers to become available...
[spawner-13] [INFO] [1742483728.147094603] [spawner_abb_joint_state_broadcaster]: waiting for service /abb/controller_manager/list_controllers to become available...
[ur5e_controller-6] [INFO] [1742483728.206365630] [ur5e.ur5e_controller]: Waiting for action server to be ready...
[abb_controller-7] [INFO] [1742483728.209441500] [abb.abb_controller]: Waiting for action server to be ready...
[webots_controller_abbirb4600-4] [ros2run]: Aborted
[ERROR] [webots_controller_abbirb4600-4]: process has died [pid 157609, exit code 250, cmd '/home/samagalhaes/ros_ws/install/webots_ros2_driver/share/webots_ros2_driver/scripts/webots-controller --robot-name=abbirb4600 --protocol=ipc --port=1234 ros2 --ros-args -r __ns:=/abb -p robot_description:=/home/samagalhaes/ros_ws/install/webots_ros2_universal_robot/share/webots_ros2_universal_robot/resource/webots_abb_description.urdf -p use_sim_time:=True -p set_robot_state_publisher:=True --params-file /home/samagalhaes/ros_ws/install/webots_ros2_universal_robot/share/webots_ros2_universal_robot/resource/ros2_control_abb_config.yaml'].
[INFO] [webots_controller_abbirb4600-4]: process started with pid [157992]
[robot_state_publisher-9] [INFO] [1742483729.374553141] [ur5e.robot_state_publisher]: Robot initialized
[webots_controller_UR5e-3] [INFO] [1742483729.387414570] [ur5e.controller_manager]: Subscribing to '/ur5e/robot_description' topic for robot description.
[webots_controller_UR5e-3] [INFO] [1742483729.390278081] [ur5e.UR5e]: Controller successfully connected to robot in Webots simulation.
[webots_controller_UR5e-3] [INFO] [1742483729.397785507] [ur5e.controller_manager]: Received robot description from topic.
[robot_state_publisher-8] [INFO] [1742483730.056241590] [abb.robot_state_publisher]: Robot initialized
[webots_controller_abbirb4600-4] [INFO] [1742483730.066122366] [abb.controller_manager]: Subscribing to '/abb/robot_description' topic for robot description.
[webots_controller_abbirb4600-4] [INFO] [1742483730.068551880] [abb.abbirb4600]: Controller successfully connected to robot in Webots simulation.
[webots_controller_abbirb4600-4] [INFO] [1742483730.076662252] [abb.controller_manager]: Received robot description from topic.
[webots_controller_UR5e-3] [INFO] [1742483730.078624832] [ur5e.UR5e.resource_manager]: Initialize hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1742483730.078788557] [ur5e.UR5e.resource_manager]: Successful initialization of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1742483730.079146107] [resource_manager]: 'configure' hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1742483730.079150516] [resource_manager]: Successful 'configure' of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1742483730.079161507] [resource_manager]: 'activate' hardware 'WebotsControl' 
[webots_controller_UR5e-3] [INFO] [1742483730.079172865] [resource_manager]: Successful 'activate' of hardware 'WebotsControl'
[webots_controller_UR5e-3] [INFO] [1742483730.079179728] [ur5e.controller_manager]: Resource Manager has been successfully initialized. Starting Controller Manager services...
[webots_controller_UR5e-3] [INFO] [1742483730.140266763] [ur5e.controller_manager]: Loading controller : 'ur_joint_state_broadcaster' of type 'joint_state_broadcaster/JointStateBroadcaster'
[webots_controller_UR5e-3] [INFO] [1742483730.140321967] [ur5e.controller_manager]: Loading controller 'ur_joint_state_broadcaster'
[webots_controller_UR5e-3] [ERROR] [1742483730.140333814] [ur5e.controller_manager]: Loader for controller 'ur_joint_state_broadcaster' (type 'joint_state_broadcaster/JointStateBroadcaster') not found.
[webots_controller_UR5e-3] [INFO] [1742483730.140385379] [ur5e.controller_manager]: Available classes:
[webots_controller_UR5e-3] [INFO] [1742483730.140409368] [ur5e.controller_manager]:   joint_trajectory_controller/JointTrajectoryController
[webots_controller_UR5e-3] [INFO] [1742483730.140909286] [ur5e.controller_manager]: Loading controller : 'ur_joint_trajectory_controller' of type 'joint_trajectory_controller/JointTrajectoryController'
[... 138 lines cut ...]
[... cut ...]
[ERROR] [abb_controller-7]: process has died [pid 157801, exit code -2, cmd '/home/samagalhaes/ros_ws/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/abb_controller --ros-args -r __ns:=/abb'].
[ERROR] [ur5e_controller-6]: process has died [pid 157800, exit code -2, cmd '/home/samagalhaes/ros_ws/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/ur5e_controller --ros-args -r __ns:=/ur5e'].
[INFO] [robot_state_publisher-8]: process has finished cleanly [pid 157802]
[INFO] [robot_state_publisher-9]: process has finished cleanly [pid 157803]
```

ralphieraccoon, 2025-03-22:
I think this may actually be caused by an earlier error:

```
what():  URDF cannot be parsed by a `ros2_control` component parser: no ros2_control tag
```

franklinselva, 2025-03-25:
> It looks like one of the recent updates to the controller manager broke this package. I rolled back the ros2_control to version 4.25.0, and everything is smooth again.

To me, downgrading ros2-control is causing segmenation fault on ubuntu 24.04. Tried both 4.25.0 and 4.26.0.

I would appreciate a quick patch or a solution to this issue.

ralphieraccoon, 2025-03-29:
I've done a little debugging, and it seems the `parse_control_resources_from_urdf` function in the `component_parser` file in `ros2_control` is receiving the wrong URDF, it's receving the robot description with all the joints and links instead of the `ros2_control` description. Will see if I can find a fix.

ralphieraccoon, 2025-04-01:
OK, so I think the first issue is down to deprecated behavior in `webots_ros2_universal_robot`, as `ros2_control` now ignores URDF files passed in as parameters and instead always reads the `\robot_description` topic. This means separate URDF files for the robot description and ros2 control controllers are no longer possible and must be combined and published on the robot description topic as is done in the other examples.

The second appears to have been fixed in ros-controls/ros2_control/#2115 so compiling the `master` branch of `ros2_control` _should_ fix it, but now I'm getting a segfault.